gumyr/build123d #19: Contextual help via LSP

smallstepman, 2022-09-26:
In cq v2 one can develop things using "dot-tab workflow", e.g. 
```python
x = cq.Workplane().box(1,2,3)
x. <PRESS TAB to open LSP hints>
```

Do you think such workflow would be possible with build123d?

gumyr, 2022-09-26:
Improving support for IDEs was one of the design goals. To facilitate expansion build123d is class based not method based so this works a little differently.

This is what I see with Visual Studio Code:
[image]
Are you looking for more than this?  It would be nice to be able to shorten this list to just the build123d specific classes - possibly even the ones that would apply in that context.

Enums are used throughout for similar reasons:
[image]
one can clearly see what the options are and quickly select one of them.

I don't know much about LSP, is there a language specification that could be created to improve usability?

smallstepman, 2022-09-30:
I'm no expert either, but I believe there is a way to sort entries in the suggestion list, however, I tried to search for it but could not find anything

gumyr, 2023-01-05:
Don't know of anything else that can be done at this point to further enhance contextual help.

closed 2023-01-05 by gumyr