gumyr/build123d #22: HexLocations: set spacing with a named parameter

roipoussiere, 2022-10-03:
An hexagon size can be defined by the size of its inner circle, or the size of its outer circle.

Depending of the use case, both can be useful for creating an hexagonal grid.

For instance, the following code makes a grid of circles with a defined spacing of 0.5mm between each.

```py
import build123d as bd

with bd.BuildPart() as builder:
    with bd.BuildSketch():
        with bd.HexLocations(2.31 + 0.5, 4, 4):
            bd.Circle(1)
    bd.Extrude(amount=1)

show_object(builder.part)
```
[image]

Here the number `2.31` can be calculated from the circle diameter, but it's tedious.

Function signature proposition:

    HexLocations(x_spacing, y_spacing, inner_diameter, outer_diameter)

And the code could be:

    with bd.HexLocations(4, 4, outer_diameter=2):

Other related suggestion:

Many functions like `Circle` use the radius, maybe the same could be applied on `HexLocations`:

    HexLocations(x_spacing, y_spacing, inner_radius outer_radius)

gumyr, 2022-10-07:
What do you think about this - a method to easily calculate the dimension?
```py
    @staticmethod
    def calc_diagonal(radius: float, spacing: float) -> float:
        """Calculate the size of the hex grid filled with circles at the given spacing"""
        return 2 * (2 * radius + spacing) / sqrt(3)
```
and in use:
```py
# Generate list of tube locations
with Workplanes("XY"):
    tube_locations = [
        l
        for l in HexLocations(
            diagonal=HexLocations.calc_diagonal(
                radius=tube_diameter / 2, spacing=tube_spacing
            ),
            x_count=exchanger_diameter // tube_diameter,
            y_count=exchanger_diameter // tube_diameter,
        )
        if l.position().Length < bundle_diameter / 2
    ]
```

The issue of using `diagonal` verse `inner_radius` or `outer_radius` is one I struggled with.  This diagonal measurement is the only non "radius" type of dimension in build123d but it seems to be a common way to specify the dimension of a hexagon.  What would you prefer as an alternative: `outer_radius`, `max_radius`, or `diagonal_radius`?

roipoussiere, 2022-10-08:
> What do you think about this - a method to easily calculate the dimension?

It's better, but a bit verbose. Why you don't like the named parameters solution?

> This diagonal measurement is the only non "radius" type of dimension in build123d but it seems to be a common way to specify the dimension of a hexagon.

Just wondering, why all circular dimensions in build123d uses the radius and not the diameter?

> What would you prefer as an alternative: outer_radius, max_radius, or diagonal_radius?

Ordered by preference: outer_radius, diagonal_radius, max_radius.

gumyr, 2022-10-08:
> It's better, but a bit verbose. Why you don't like the named parameters solution?

In this case there would be two parameters that are mathematically fixed together which doesn't seem optimal in that the user could submit contradicting values.

> Just wondering, why all circular dimensions in build123d uses the radius and not the diameter?

Fillets don't have a diameter only a radius so it would cause the exception to the rule of using either radius or diameter.

Thinking about this some more, how about replacing `diagonal` with `apothem` (the radius of the inscribed circle) or maybe `inscribed_circle_radius`? If the user wants to add space between the circles they just need to add 1/2 of that space to this `apothem` parameter. Therefore, only one parameter that is easy to use if one is packing circles.

roipoussiere, 2022-10-12:
Ok for the first two points.

> Thinking about this some more, how about replacing diagonal with apothem (the radius of the inscribed circle) or maybe inscribed_circle_radius?

I don't understand this geometrical relation.

Maybe a flag could suffice? Like:

```py
HexLocations(radius: float, x_spacing: float, y_spacing: float, use_inner_radius: bool)
```

closed 2023-01-27 by gumyr