gumyr/build123d #26: bad IDE integration

roipoussiere, 2022-10-10:
Some classes are incorrectly recognized as modules:

[image]

Also, auto-completion doesn't work with enumerations:
[image]

I must type `bd.Mode.Mode` to have a correct auto-completion:
[image]

gumyr, 2022-10-10:
I know this generally isn't the best python practice but when imported as `from build123d import *` all of these problems go away (at least in my environment).  It will take some research to figure out why alternative imports cause this issue.

gumyr, 2023-02-12:
With commit 2b830ea472b267d41e3b6878d181ec679e2e6483, this problem seems to be solved. The issue was caused by importing from the top level of the project instead of from the appropriate module.

[image]

[image]

[image]

closed 2023-02-12 by gumyr