gumyr/build123d #35: Isn't the "about" description of the project somewhat misleading?

adam-urbanczyk, 2022-10-18:


gumyr, 2022-10-19:
Possibly. What specifically do you find misleading?

adam-urbanczyk, 2022-10-19:
In what sense is this repo a prototype of cadquery 3 and what is cadquery 3 actually?

gumyr, 2022-10-19:
Dave Cowden has lead a community effort to define the roadmap of CadQuery - see here: https://github.com/CadQuery/cadquery/wiki/RoadMap.  The CadQuery community on Discord has spent a lot of time and effort in discussing and prototyping what that next step might look like and build123d is the only proposal that has achieved substantial functionality. 

You are welcome to contribute to the discussion or to build123d itself.

adam-urbanczyk, 2022-10-20:
@dcowden @jmwright is this project recognized by you as an official "cadquery 3" proto?

dcowden, 2022-10-20:
@gumyr @adam-urbanczyk Build123d incorporates a lot of ideas from discussion in the cadquery forums. 
 
Roger's notes in discord on 10/16 :

"One of my goals is to have no dependencies on cadquery once I'm done which will allow for a proper release as well as avoiding the user having to create two complex environments. Assembly and the importer/exporters still need to be ported. I'm trying to do this an incrementally as I can."

I can't see how build123d can be a cadquery prototype with no dependencies on cadquery.

gumyr, 2022-10-20:
I'll remove any mention of "cadquery prototype" - it seems that the only place that's mentioned is in the pip package info - was there somewhere else?

dcowden, 2022-10-20:
@gumyr I dont know of other places. 

Also, I think its fine to retain a connection with cadquery more generally.  More specifically, continuing the development of build123d using the cq discord and community.  You're building a lot of functionality with community involvement, which is great. I'd be right there with you if I had the bandwidth. I wouldn't be surprised if CQ eventually incorporates many of the ideas, especially the ones people really like. 

The main difference between CQ 3 and build123d is that CQ3's topmost api layer ( fluent, builder, whatever) will certainly contain dependencies on the cq direct api, as is mentioned in the Roadmap.

gumyr, 2022-10-21:
The build123d top layer is dependent on a revamped version of the cadquery direct api - see here: https://github.com/gumyr/build123d/blob/dev/src/build123d/direct_api.py  What I've done here is primarily:
- make it PEP8 compliant
- replace the use of Literal strings with Enum classes
- introduce an `Axis` class
- introduce a `ShapeList` allowing the direct api to use "selectors"
- improve consistency across all of the api (e.g. degrees everywhere, consistent parameter names, Vertex construction following the Vector model, no more `faces()` and `Faces()`, etc.)
- I expect there will be a few others as well

However, this is fundamentally the same direct api that cadquery is based on - method based, sames classes, etc. The scope of the changes do break backwards compatibility - however, it certainly would be possible to refactor `Workplane` and `Sketch` to use it if there was a need - so that is why I'm trying to create something that can be independently installed (one would expect that cq 3 would not be dependent on cq 2 or cq 1).

If nothing else, hopefully this "prototype" illustrates some of what could be done.

dcowden, 2022-10-21:
sounds good. I think its' reached > prototype status, looks like you have some folks using it, which is fantastic!  I'm very interested to see how it grows.  That experience will do very well to inform future CQ direction

adam-urbanczyk, 2022-10-21:
Great, please update this and then the issue can be closed.

[image]

closed 2022-10-22 by gumyr