gumyr/build123d #38: CenterOf.GEOMETRY seems to return non correct center values

bernhard-42, 2022-10-23:
I have a simple chamfered box:

```python
with BuildPart() as b:
    Box(1, 2, 3)
    Chamfer(*b.edges(), length=0.2)
```

To my surprise, `group_by` delivers only 6 of 8 faces. 
Reasons seems to be `CenterOf.GEOMETRY`: 

`[o.center(CenterOf.BOUNDING_BOX).Z for o in b.faces().sort_by(Axis.Z)[-9:]]` returns `[1.4, 1.4, 1.4, 1.4, 1.4, 1.4, 1.4, 1.4, 1.5]`, i.e. all except the top face have z-value 1.4

`[o.center(CenterOf.GEOMETRY).Z for o in b.faces().sort_by(Axis.Z)[-9:]]` returns `[1.35, 1.35, 1.4, 1.4, 1.4, 1.4, 1.4, 1.4, 1.5]`, i.e. with the default geometry center, two of the triangles have a smaller z-value than the other two.

gumyr, 2022-11-28:
With your PR, can this issue be closed now?

bernhard-42, 2022-11-28:
The situation is better, I now get 4 chamfered rectangles with `[-2]` and the 4 chamfered triangles with `[-3]`
If I could provide the center param to group_by, I could use BB which would give me back all 8 rectangles and triangles.

Independent of this use case, given the sensibility of center, I would love to see sort_by and group_by getting an additional parameter `center_of` with GEOMETRY as default.
What do you think?

gumyr, 2023-08-21:
`center` is documented here: https://build123d.readthedocs.io/en/latest/center.html

closed 2023-08-21 by gumyr