gumyr/build123d #67: export_svg 'Compound has no attribute center'

jdegenstein, 2022-11-30:
When trying to run export_svg from within CQ-editor I get the following error. (Note that running `python script.py` works fine):
```py
three_d.part.export_svg(
        "three_d.svg",
        (-1, -.5, 200),
        (0, 0, 1),
        svg_opts={
            "pixel_scale": 20,
            "show_axes": False,
            "show_hidden": False,
        },
    )
```
[image]

Also, no SVG is output when this error occurs.

gumyr, 2022-12-01:
The last build123d commit (6cf4b0428fdd0c093b0b823f36e2137fc6498e6b) fixed this problem.
```py
from build123d import *

c = Compound.make_compound([Solid.make_box(1, 1, 1)])
print(c.center())
```
```
Vector: (0.5, 0.5, 0.5)
```
Does the export work if you upgrade to the latest?

jdegenstein, 2022-12-01:
You were right, sorry I missed that in the changelog! Everything works great now -- thank you. I will close the issue.