gumyr/build123d #75: Separate projection from `export_svg`

gumyr, 2022-12-06:
When exporting to SVG a projection is done of the Shape to create a set of visible and hidden Edges. These generated Edges are written to the SVG file.  As such the projection is a general operation not specific to SVG. In addition, separation would potentially allow more control - when color is added to Shape objects (and potentially stroke width & stroke style to Edge/Wires) these attributes could be used directly when writing to SVG (or DXF?).

`export_svg` would then expect planar Edges/Wires and be able to translate them to the XY plane for writing.

jrmobley, 2023-03-16:
You can assign this to me.

jrmobley, 2023-03-19:
Making progress.  Separated the projection into a `Drawing` class and rebuilt the SVG export to export true circles and Bézier paths.  Also has multi-layer support similar to the latest CQ DXF exporter.  Plenty of little details still to work out, and then refactoring DXF into the system.

A projected drawing on an enclosure:
[image]

And a planar section of the matching PCB.
[image]

Paul8043, 2023-05-04:
@jrmobley 
This is a very good decision. The projection should be separated from the exporter. So, projection is an optional step before the exporter has been called. This helps a lot for applications that want to pack a 2D-objects into a SVG-file. My typical workflow is to design a 3D-model and then make a section() on a certain plane and the result should go into a 2D-SVG-file. This is exactly what is needed for my [router](https://www.shapertools.com/en-us/origin). The SVG-file will directly drive this router without any cam-application in between. The color-encoding (see [cut-type-encoding](https://support.shapertools.com/hc/en-us/articles/115002721473-Cut-Type-Encoding)) determines what the router should perform. My focus here is the 2D-SVG-exporter, it should preserve the shapes, e.g. circles should be encoded as circles, and not as polylines. And it is important that polylines are closed, this will enable some very comfortable functions inside the router (helix-mode). And, also very important, the SVG-exporter must do its job without any internal re-scalings, the measures must be preserved in any case. I would like to discuss some details with you, just to make sure that the 2D-SVG-exporter becomes an interface that is universal enough to accomplish such requirements. Maybe we can make close cooperation. Let me know, if this fits to your plans and activities.

gumyr, 2023-05-05:
I've only done a little non-functional work on the code to start to get it to pass pylint. It will need unit testing, examples and documentation as well so there is a fair amount still to do.

Paul8043, 2023-05-09:
I still need a few days to port a cadquery-project to build123d. Coding with build123d makes fun. I like the architecture of build123d, especially the algebra-mode. Great work, thanks for this contribution. Let me know, when the projection-free SVG-exporter will be available.

gumyr, 2023-08-20:
Done with commit 76c4f2e

closed 2023-08-20 by gumyr