gumyr/build123d #77: Performance Enhancement: Clean at the end of active locations

jdegenstein, 2022-12-13:
Clean is called every time an object is added, in the test1 case below it is called ~255 times. One idea proposed was to consider moving clean to the end of the active locations so that it would only be called once in the test1 case below:

```py
from build123d import *
import time

diam   = 175
meshop = 2
gridxy = 16 #int(diam/meshop/2)

a = time.time()

#TEST1
 with BuildPart():
     with BuildSketch() as holes1:
         with GridLocations(meshop*2,meshop*2,gridxy,gridxy):
             Rectangle(meshop,meshop)
     with BuildSketch() as holes:
         Add(holes1.sketch)
         #Circle(diam/2,mode=Mode.INTERSECT)
#13.44 seconds

#TEST2
#with BuildPart():
#    with BuildSketch() as holes:
#        with GridLocations(meshop*2,meshop*2,gridxy+1,1):
#            Rectangle(meshop,diam)
#        with GridLocations(meshop*2,meshop*2,1,gridxy+1):
#            Rectangle(diam,meshop)
#        #Circle(diam/2, mode=Mode.INTERSECT)
#2.41 seconds

if "log" and "show_object" in locals():
    log(time.time() - a)
    show_object(holes.sketch,options=rand_color())
else:
    print(time.time() - a)
```

jdegenstein, 2022-12-16:
One easy fix from gumyr: In BuildSketch Rectangle change this:
```py
        for face in new_faces:
            context._add_to_context(face, mode=mode)
```
to this:
```py
        context._add_to_context(*new_faces, mode=mode)
```
This ONLY fixes BuildSketch Rectangle and there are a dozen or so instances of the slower version in other geometries. The faster version only takes `0.21 seconds` to run compared to `13.44 seconds` above.

gumyr, 2023-01-27:
f935352ac02a558ee7a25580f58ede43c7d46f18 should fix this problem.  @jdegenstein, would you profile it and check?  Thanks in advance.

jdegenstein, 2023-02-01:
With the latest commit, this has had a huge performance increase. On the same machine as before, "TEST1" completed in `0.135 seconds` (around 100x faster, and probably higher due to other overhead), and "TEST2" in `0.43 seconds`.

closed 2023-02-01 by jdegenstein