gumyr/build123d #91: extrude_until fails when face and target_object have same width

bernhard-42, 2023-01-07:
This is an interesting test case since it is pretty simple (similar to the one that you've shown when you introduced the new `extrude_until`:

```python
face_width = 0.9999
diameter = 1.0

c = Edge.make_three_point_arc((0, 0, -5), (0, -5, 0), (0, 5, 3))
f = Face.make_from_wires(Wire.make_circle(diameter/2)).locate(
    Plane.XZ.to_location() * Location((0, -5, 0))
)
t = Solid.sweep(f, Wire.make_wire([c]))

f2 = Face.make_rect(8, face_width).locate(
    Plane.XZ.to_location() * Location((0, 0, -7))
)

p = Solid.extrude_until(f2, t, (0, -1, 0))
```

It works when `face_width` is slightly smaller then `diameter` and fails when they are the same (1.0).

The reason is that in case of `face_width == diameter` some of the cuts or intersects with `clipping_object` run into the exception catching of `Solid.extrude_until`

This also happens for `Until.NEXT` and `Until.LAST`

gumyr, 2023-01-07:
I'm not sure if anything can be done to fix this.  Here is the clipping object that is causing the trouble: 
[image]
which looks fine but because the face being extruded has tangents that are equal to the direction of extrusion I suspect the resulting clipping object is not really valid.

I've updated the docstring saying the bounding surface must be larger and added a warning to help users cope with this limitation (see commit 16ececb36d0de5dc89510694eb7c3168fd213476).  Is there anything else that you think could be done?

bernhard-42, 2023-01-08:
I took it as an opportunity to better understand some more OCCT concepts and used the hidden line removal algorithm (HLR*)  to project the clipping faces onto a plane, in order to split the section into pieces that I then can then simply extrude linearly.
While it worked (with some location things still unclear to me), for "difficult" objects (i.e. artefacts when the section is as wide as the target object), some of the following extrusions of these section parts failed to produce proper solids.

Another indication is that `extrusion.cut(target_object)` doesn't work when section and target object are of same width.

Hence, I agree with your approach to add the warning to the docs. And I like the warnings. They make it clear that something did not go as expected.

I guess, this issue can be closed.

bernhard-42, 2023-01-08:
There are some more issues, even when sticking to the rules:

```python
c = Face.make_from_wires(Wire.make_circle(2)).located(Location((20, 0, 0)))
a = Solid.revolve(c, -180, Axis.Y)
f = Face.make_rect(3.9, 20)
e = Solid.extrude_until(f, a, (0, 0, 1))
```

[image]

The face is 0.1 less wide than the arc, however the result is:
[image]

One issue here is that the seam of the torus arc is at the top. This means that there are two extrusions at the top when calculating the `clipping_objects` in the routine and these two clipping objects are invalid:
```python
clipping_objects[0].is_valid()
True
clipping_objects[1].is_valid()
False
clipping_objects[2].is_valid()
False
```

This can be handled by adding `.fix()` here:

```python
        clipping_objects = [
            Solid.extrude_linear(f, clipping_direction).fix() for f in clip_faces
        ]
```
Afterwards the clipping objects are valid.

However, cutting
[image]

from 

[image]

Doesn't seem to do anything (despite the boolean operation return True for `IsDone()`.

The next cut 

[image]

produces a bad result:

[image]

bernhard-42, 2023-01-08:
Something is wrong with `cut`:

[image]

This should nicely cut the upper part (clipping_objects[0]) from the lower part (`extrusion` without subtracting `target_object`- just to make it easier). However, `extrusion.cut(clipping_objects[0])` returns
[image]

The culprit lies in `_bool_op`:
If I show `operation.Shape()`,  the boolean operation actually has created a shape with two solids (no idea why):

[image]

And the last call in `_bool_op` (`Shape.cast`) combines them to one solid having both parts
[image]

bernhard-42, 2023-01-08:
`operation.Shape.NbChildren()` correctly returns `2`.
I don't know how to continue from here. Is the first element always the one we want?
Or did I miss something?
Any idea?

gumyr, 2023-01-14:
I'm still looking at Until.LAST but for Until.NEXT, the `fix()` change and:
```py
                extrusion = (
                    extrusion.cut(clipping_object).solids().sort_by(direction_axis)[0]
                )
```
I get:
[image]

It seems like Until.LAST should always be Until.NEXT plus the intersection of the extruded face and the part.  Unfortunatley, I'm getting:
[image]

Jojain, 2023-01-14:
I think that cases where the face to extrude is wider or the same size of the face it will intersect will be hard to solve. I think that's ok if that's a limitation

gumyr, 2023-01-14:
Commit 1fb55a299e1a68cb87d21d23e0b8d4e9e934285c has this minor improvement

gumyr, 2023-01-16:
Applied same minor improvement to Until.LAST case - commit 2a22cf70f9666733c29bd59c03a6591f810dc792

Although there may be geometries that will cause extrude_until to fail - I don't see any way to improve it at this point and therefore will close this issue.  Thanks for all the work done here to improve it!

closed 2023-01-16 by gumyr