gumyr/build123d #95: Difference in rotation of RegularPolygon and PolarLocations with odd # sides

gumyr, 2023-01-08:
```py
    RegularPolygon(100, 7)
    with PolarLocations(55, 7):
```
doesn't line up very well, requiring:
```py
    RegularPolygon(100, 7, rotation=-360 / 28)
    with PolarLocations(55, 7):
```
to be useful.

bernhard-42, 2023-01-11:
Additionally , `RegularPolygon` has its normals pointing into -z direction, unlike the other sketch objects
[image]

bernhard-42, 2023-01-11:
For reference:

```python
with BuildSketch() as h:
    RegularPolygon(2,7)
    with Locations(Location((4,0,0))):
        Rectangle(2,2)
    with Locations(Location((0,4,0))):
        Circle(2)
    with Locations(Location((0,-4,0))):
        Ellipse(2, 1.5)
```

jdegenstein, 2023-01-11:
Possibly related behavior in this example:
```py
from build123d import *

length, width, thickness = 80.0, 60.0, 10.0

with BuildPart() as ex19:
    with BuildSketch() as ex19_sk1:
        RegularPolygon(radius=length/2, side_count=5)
    Extrude(amount=thickness)
    topf = ex19.faces().sort_by(Axis.Z)[-1]
    with Workplanes(topf) as zz:
        log(zz.workplanes)
```
Returns: `[Plane(o=(0.00, 0.00, 10.00), x=(-1.00, 0.00, 0.00), z=(-0.00, -0.00, 1.00))]` Note that this ALWAYS returns the negative x-dir including with even/odd number of sides.

ezrec, 2023-01-14:
As the PR has merged, this issue can be closed, correct?

gumyr, 2023-01-14:
Before the issue is closed I'd like to see a test case added to cover the problems.  The location of the vertices of `RegularPolygon` must be the same as that of `PolarLocations` and the normal of `RegularPolygon` must be +Z. @ezrec, if you're willing I be grateful for a pull request with the new test.

closed 2023-01-16 by gumyr