gumyr/build123d #103: Create BREP shapes from mesh?

gumyr, 2023-01-18:
From @greyltc

As others have said, you might use a 3d scanner to get a mesh from a real object. I've had decent success going from mesh -> solid model using FreeCAD. Import the mesh, choose the part toolbox and go "Create Shape from mesh..." --> "Convert to solid" --> "Create a copy" --> "Refine shape." You might get something from that workflow that's surprisingly good!

jdegenstein, 2023-01-24:
I was curious about this in the context of OCCT, and this discussion from around 6 years ago seems to indicate it was not possible with OCCT. https://dev.opencascade.org/content/convert-triangular-mesh-brep

Jojain, 2023-01-24:
That's a tough problem that has already been an issue for topological optimization. Current implementation requires the user to rework upstream it's design given the result of the optimization. Probably an area where some kind of IA could do something to improve.

Now I don't think it's worth loosing time on this problem. I think there is more solvable issues that should have the priority

greyltc, 2023-01-24:
[Rabbit.stl.zip](https://github.com/gumyr/build123d/files/10494255/Rabbit.stl.zip)
[RabbitSolid.step.zip](https://github.com/gumyr/build123d/files/10494271/RabbitSolid.step.zip)
I've attached a rabbit-mesh (.stl) that I turned into a rabbit-solid (.step) with the workflow above.

gumyr, 2023-01-25:
I think a little more could be done than what is done here (there is an stl import here: https://github.com/JustinSDK/cqMore).
[image]

This method (https://build123d.readthedocs.io/en/latest/direct_api_reference.html#direct_api.Face.sew_faces) allow faces to be "sewed" together or unified.  If one was to go through each of the faces in the rabbit and find the faces with normals "almost" in the same direction - they could be rebuilt as non-planar by reusing the three edges and a calculated central vertex, then sewn together. If that doesn't take forever it might be interesting.

Jojain, 2023-01-26:
It sure seems interesting, it will obviously take a while and the result will be more or less usable depending on how smart is the Algo to merge faces, the real problem honestly is that stl is the standard for 3D printing but it's such a lossy format. Especially when most of these files are made in software using BREP that's a shame  that these files are not shared as STEP, it would solve a lot of issues !

gumyr, 2023-08-21:
Currently the Mesher class (see https://build123d.readthedocs.io/en/latest/import_export.html#d-mesh-import) allow both STL and 3MF (mesh) files to be imported as build123d Solid objects. However, no further processing as described above is done to attempt to recover the "true" BREP shape which therefore results in a faceted object.

AnicetNgrt, 2024-08-11:
Some interesting papers using ML to reconstruct brep from stl or point cloud data
https://arxiv.org/abs/2105.09492
https://arxiv.org/abs/2401.15563

jdegenstein, 2026-08-18:
Update: @gumyr added https://github.com/gumyr/build123d/blob/dev/src/build123d/brep_from_stl.py

Currently the above can detect and fit planar, cylindrical and spherical faces. It is my understanding that these are untrimmed faces and do not enclose a proper BREP solid. To that end I have been investigating OCCT's `BOPAlgo_MakerVolume` for awhile. This can take in a set of untrimmed faces and find the volume(s) bounded by these faces. Here is an example screenshot using my implementation (below) of a function called `make_volume`: 

[image]

```py
cyl = Cylinder(2,15).faces().filter_by(GeomType.CYLINDER)
cyl2 = Cylinder(1,15).faces().filter_by(GeomType.CYLINDER)
rect = Pos(Z=-4)*Rectangle(10,10)
rect2 = Pos(Z=4)*Rectangle(10,10)
faces = []
faces.extend(cyl)
faces.extend(cyl2)
faces.extend(rect)
faces.extend(rect2)
result = Pos(X=10,Y=10)*make_volume(faces)
```

The exciting thing here is that this means it is now theoretically possible to take an appropriate mesh (composed only of planar, cylindrical, spherical faces) and feed it through @gumyr 's new brep_from_stl methods and then convert that into a true BREP solid with my implementation of `make_volume`. I have not tested it end to end yet, but here is my implementation of `make_volume`:

```py
from typing import Iterable
from OCP.BOPAlgo import BOPAlgo_MakerVolume
from OCP.TopTools import TopTools_ListOfShape
from build123d import Part, Face

def make_volume(
    faces: Face | Iterable[Face],
    intersect: bool = True,
    avoid_internal_shapes: bool = True,
    fuzzy_value: float = 0.0,
) -> Part:
    """Builds solid volume cells from bounding/intersecting faces using BOPAlgo_MakerVolume.

    Args:
        faces: A single Face or an iterable of Faces.
        intersect: If True, computes mutual intersections between faces.
        avoid_internal_shapes: If True, discards non-boundary internal shapes.
        fuzzy_value: Additional tolerance for fuzzy Boolean intersections.

    Returns:
        A build123d Solid (if exactly one volume is created) or a Compound of Solids.

    Raises:
        RuntimeError: If BOPAlgo_MakerVolume fails or produces no shapes.
    """
    face_list = [faces] if isinstance(faces, Face) else list(faces)

    if not face_list:
        raise ValueError("At least one Face must be provided to make_volume.")

    # Populate the OCCT argument list
    shape_list = TopTools_ListOfShape()
    for face in face_list:
        shape_list.Append(face.wrapped)

    # Configure and run BOPAlgo_MakerVolume
    maker = BOPAlgo_MakerVolume()
    maker.SetArguments(shape_list)
    maker.SetIntersect(intersect)
    maker.SetAvoidInternalShapes(avoid_internal_shapes)
    maker.SetRunParallel(True)

    if fuzzy_value > 0.0:
        maker.SetFuzzyValue(fuzzy_value)

    maker.Perform()

    if maker.HasErrors():
        raise RuntimeError("BOPAlgo_MakerVolume failed during volume generation.")

    result_shape = maker.Shape()
    if result_shape.IsNull():
        raise RuntimeError("BOPAlgo_MakerVolume generated a null shape.")

    return Part(result_shape)
```