gumyr/build123d #110: No good way to select faces by relative orentation.

ezrec, 2023-02-01:
The 'sort_by()' method of a set of faces always uses the global coordinate system; making face selection in deeply nested Workplanes() problematic. With 90 degree workplanes this can be 'mathed' out, but with workplanes that are on arbitrarily rotated faces (for example, adding a complex workplane operation to each face for a dodecahedron) this quickly gets out of hand.

Trivial example:

```py
        with BuildPart() as result:
            b1 = Box(20, 30, 40)
            with Workplanes(b1.faces().sort_by(Axis.X)[-1]):
                b2 = Box(10, 10, 20)
                # User assumes they can select the _relative_ top Z face of b2 in this context:
                with Workplanes(b2.faces().sort_by(Axis.Z)[-1]):
                    Hole(2, 10)
```

Expected result:

[image]

Actual result:

[image]

<hr>

Suggested _non breaking_ syntax change:
- Extend the existing `Workplane.localize()` method to handle `Axis` classes, that translates an Axis into the relative coordinate space of the Workplane

```py
        with BuildPart() as result:
            b1 = Box(20, 30, 40)
            with Workplanes(b1.faces().sort_by(Axis.X)[-1]) as wp:
                b2 = Box(10, 10, 20)
                # User _specifies_ they select the _relative_ top Z face of b2 in this context:
                local_z = wp.localize(Axis.Z)
                with Workplanes(b2.faces().sort_by(local_z)[-1]):
                    Hole(2, 10)
```

ezrec, 2023-02-01:
(Well, what I'd _really_ like would be to have `with Workplanes()` push a new global coordinate system on the context for all underlying direct API functions to use but that would be (a) an API breaking change and (b) a ton of work to validate)

gumyr, 2023-02-01:
This is an interesting perspective, thanks for bringing it to my attention.

You are correct in saying that "`sort_by()` method of a set of faces always uses the global coordinate system", this is by design.  One can achieve the functionality that you're looking for by localizing the sort direction Axis as shown in this alternative to your design:

```py
with BuildPart() as alternative:
    b1 = Box(20, 30, 40)
    # Create a new workplane from a selected face
    workplane = Plane(b1.faces().sort_by(Axis.X)[-1])
    # Create a z_axis relative to this new workplane
    workplane_z_axis = Axis(workplane.origin, workplane.z_dir)
    with Workplanes(workplane):
        b2 = Box(10, 10, 20)
        with Workplanes(b2.faces().sort_by(workplane_z_axis)[-1]):
            Hole(2, 10)
```
Your ideal solution would only be practical if `Workplanes` only accepted a single `Plane`. I don't see how it could work practically if in your example if at step `b2` two workplanes (say the front and back) were active.

At this point I don't see any way to further optimize this design style - is there something I'm missing that you'd like to see?

ezrec, 2023-02-01:
Ah yes, I see now. I had a fundamental misunderstanding of what was going on.

I had internalized the syntax of `with Workplanes(a, b, c...) as wp` incorrectly as `for wp in Workplanes(a, b, c...)`, which is _not_ semantically equivalent.

closed 2023-02-01 by ezrec