gumyr/build123d #135: Idea: Operation mode context

hiaselhans, 2023-02-14:
given the following example (from examples/multiple_workplanes.py):

from build123d import *

``` python
with BuildPart() as obj:
    Box(5, 5, 1)
    with Workplanes(*obj.faces().filter_by(Axis.Z)):
        Sphere(1.8, mode=Mode.SUBTRACT)
```

What if we took the context-manager approach one step further and were enabling something like this:

from build123d import *

``` python
with BuildPart() as obj:
    Box(5, 5, 1)

    with obj.SUBSTRACT():
        with Workplanes(*obj.faces().filter_by(Axis.Z)):
            Sphere(1.8)
```

Jojain, 2023-02-14:
What does this bring ?

The context managers while being powerful are nonetheless a bit complex so I would not be in favor of adding more context managers to allow different ways of doing the same thing.

There is already so much way one can build a part, the more restrictive the API is the easiest it will be to debug.

jdegenstein, 2023-02-14:
This is not really adding anything new from an end user perspective. Consider the following re-write of the 2nd example using already existing methods:
```py
with BuildPart() as obj_sub:
    with Workplanes(*obj.faces().filter_by(Axis.Z)):
        Sphere(1.8)

with BuildPart() as obj:
    Box(5, 5, 1)
    Add(obj_sub.part,mode=Mode.SUBTRACT)
```

hiaselhans, 2023-02-14:
thanks @jdegenstein that seems decent enough for me.

Here is the same thing with nested BuildParts (in order to have obj.faces available):

``` python
with BuildPart() as obj:
    Box(5, 5, 1)

    with BuildPart() as obj_sub:
        with Workplanes(*obj.faces().filter_by(Axis.Z)):
            Sphere(1.8)

    Add(obj_sub.part,mode=Mode.SUBTRACT)
```

closed 2023-02-14 by hiaselhans

jdegenstein, 2023-02-14:
Good point, here is another tweak: `with BuildPart() as obj_sub:` should be `with BuildPart(mode=Mode.PRIVATE) as obj_sub:`  as it prevents from adding to the parent