gumyr/build123d #139: Offset should not require a minimum of 2 edges

jdegenstein, 2023-02-16:
https://github.com/gumyr/build123d/blob/7eb673a0fe10fbcebfc739708a7604fcc17b5a6b/src/build123d/build_generic.py#L419-L420

Offset raises a `ValueError("At least two edges are required")` when one edge is passed in. I see no reason for this limitation. I tested without it and it worked fine on a single edge.

Offset of a single straight edge results in a rectangle with rounded ends, which is desirable because it is similar to the concept of "stroke path" in Inkscape.

gumyr, 2023-02-17:
This was an OCCT bug - that sounds like it's been fixed - thankfully.

gumyr, 2023-02-20:
The fundamental OCCT issue is still there - a single linear Edge does not work.  However, the single edge can be broken in two and combined in a Wire and it works. ¯\_(ツ)_/¯

Fixed with latest commit.

closed 2023-02-20 by gumyr