gumyr/build123d #141: private builder mode

hiaselhans, 2023-02-19:
I tried to split a builder into pieces by building new compound classes and using them at the desired locations.
The docs advice to use a private builder context and in that context i would expect to start with a blank location (set at origin). Instead Locations from an outer scope shadow into the new context.

Here is a minimal example:

``` python
def nut():
    with bd.BuildPart(mode=bd.Mode.PRIVATE) as part:
        with bd.BuildSketch():
            bd.RegularPolygon(4, 6)
        
        bd.Extrude(amount=5)
    
    return part.part

with bd.BuildPart() as part:
    with bd.Locations((-10,-10), (10,10), (-10,10), (10,-10)) as locations:
        bd.Add(nut())

part
```
expected result:
[image]

actual result:
[image]

gumyr, 2023-02-19:
The global Locations in BuildPart are interacting with the code in the function.  This:
```py
def nut():
    with bd.BuildPart(bd.Plane.XY, mode=bd.Mode.PRIVATE) as part:
        with bd.BuildSketch():
            bd.RegularPolygon(4, 6)

        bd.Extrude(amount=5)

    return part.part
```
produces this:
[image]
Note that I'm setting the workplane and therefore the default Location.

I've worked on trying to automate this for quite a while and been unable to come up with a solution.  The requirement to set the workplane and mode in functions and classes will be documented here: https://build123d.readthedocs.io/en/latest/custom.html when I get a chance.

gumyr, 2023-02-25:
Fixed with commit 0ae15a463be547ab73fa24559d61602e7c4a6e3e

closed 2023-02-25 by gumyr