gumyr/build123d #151: Update child part positions when parent Joint position is changed

gumyr, 2023-02-26:
When many joints are used to connect many parts together, it should be possible to update a joint position and have the parts connected to this joint reposition themselves.  For example, an industrial robot with many segment rotates its base - all the other parts should move appropriately.

bernhard-42, 2023-02-26:
I would say that this is a feature for hierarchical assemblies. It is exactly how cadquery-massembly works today. Each partgroup at each level of the assembly hierarchy has its own location, By changing a partgroups location, the whole subtree moves/rotates.
Hexapod works like that. When you rotate the upper leg around the base, the lower leg follows, since the lower leg is hierarchically below the upper leg.

The question is, whether we could introduce a function "assemble" that works similar to "connect", but creates and assembly instead of a compound?

gumyr, 2023-02-28:
I've been thinking about the overlap in functionality between connected joints and assemblies.  The `connect_to` method could also be building an assembly by assigning children/parent; however, it would have to be working with a Compound to start with or even "upgrade" the Shape to a Compound. 

What if there was an option to `connect_to` to flag that an assembly should be built? Should building an assembly be fully automatic?

bernhard-42, 2023-02-28:
I don't think we need a separate method, a flag would be good.

What do you mean with "automatic"?

Another task that assembly building would need to do if we want to move parts and hierarchies is to relocate every object in the hierarchy so that its origin and rotation is aligned with the joint around which it has its dof. Because then rotate about z or move y units in joint direction could be given in local coords (i.e. in joint degrees or units  relative to the hierarchy it wants to move) which is very intuitive

gumyr, 2023-02-28:
By "automatic" I mean then when a user enters `Solid_1.connect_to(Solid_2, ...)`:

1. `Solid_1` is converted to a Compound e.g. `Compound_1`
2. `Solid_2` is relocated relative to `Compound_1`
3. `Compound_1` is assigned as `Solid_2`'s parent (i.e. creating an assembly)

As this impacts `Solid_1` a flag allowing user control of this behaviour seems appropriate.

> Another task that assembly building would need to do if we want to move parts and hierarchies is to relocate every object in the hierarchy so that its origin and rotation is aligned with the joint [...]

I believe that just re-evaluating the `connect_to` of each child part/joint would achieve this.

bernhard-42, 2023-02-28:
Understood. Without automation the difference of an "atomic" object (i.e. Solid) and a fused/cut object (i.e. Compound) needs to be understood. It is the difference between `assembly = Compound(solid).connect_to ...`  and `assembly = compound.connect_to ...`.

> I believe that just re-evaluating the connect_to of each child part/joint would achieve this.

Not sure I understand the implication of your described step. Let me explain what I did in MAssembly and then you can see whether we actually mean the same. In MAssembly this is the location handling:
- I took over from cq.Assembly that locations are hierarchical, i.e. the world location of an object in the assembly hierarchy is calculated by multiplying the locations of all objects in the hierarchy above the selected object.
- Additionally, to allow using the mates (your joints) in a natural way I relocate every object so that the origin of the object is aligned with the location of the mate that controls the movement of the object (let's call it "primary mate"): `obj.moved(primary_mate.inverse())` and do the same for all mates attached to this obj. Note, the primary mate location is now the identity (`Location()`). 
- Then I assemble the relocated objects.

This approach ensures that when you rotate an object e.g. around the z-axis, all objects below in the sub hierarchy automatically follow that movement. They are actually grouped together. threejs has implemented the exact same approach, btw.

Note that the relocation step (the second bullet) to the primary mate is only needed when you plan to animate the assembly along the mates/joints. Without animation, this step can be omitted.

It would be great if build123d would already add this step for future animation needs.