gumyr/build123d #156: Location Doubling Plane Height and Other Bad Behavior

jdegenstein, 2023-03-02:
Apologies if this is a duplicate of another issue. I am seeing two related issues with location contexts:
```py
from build123d import *
from copy import copy

def txt(strr, tloc, xoff = 0):
    iloc = copy(tloc) #needed otherwise location bleeds over
    iloc.position = iloc.position + Vector(xoff,0,0)
    with BuildPart(Plane.XY, mode=Mode.PRIVATE) as p:
        with BuildSketch() as sk:
            with Locations(iloc):
                Text(strr,
                     fontsize=3,
                     font_style=FontStyle.BOLD,
                     align=(Align.CENTER,Align.CENTER),
                     )
        Extrude(amount=1)
    return p.part

z_height = 2
with BuildPart(Plane(origin=(0,0,z_height)),mode=Mode.SUBTRACT) as zzz:
    txtlocs = GridLocations(7,0,7,1)
    txtlbl = ["-3","-2","-1","0","+1","+2","+3"]
    for idx, obj in enumerate(txtlocs):
        Add(txt(txtlbl[idx], obj))
        if idx == 0:
            Add(txt("%",obj,-3))

log(zzz.part.bounding_box().min.Z) # returns --> INFO: 3.9999999
show_object(zzz.part)
```
[image]
The z_height of the resulting objects is doubled. `log(zzz.part.bounding_box().min.Z) # returns --> INFO: 3.9999999`

The next post will cover a related problem, but keeping it separate for clarity.

jdegenstein, 2023-03-02:
If we now remove the use of copy.copy inside the function `txt` then repeated calls to a location are modified globally. I am not sure this is a bug, but it did catch me off guard.
```py
from build123d import *
from copy import copy

def txt(strr, tloc, xoff = 0):
    #iloc = copy(tloc) #needed otherwise location bleeds over
    tloc.position = tloc.position + Vector(xoff,-5,0)
    with BuildPart(Plane.XY, mode=Mode.PRIVATE) as p:
        with BuildSketch() as sk:
            with Locations(tloc):
                Text(strr,
                     fontsize=3,
                     font_style=FontStyle.BOLD,
                     align=(Align.CENTER,Align.CENTER),
                     )
        Extrude(amount=1)
    return p.part

z_height = 2
with BuildPart(Plane(origin=(0,0,z_height)),mode=Mode.SUBTRACT) as zzz:
    txtlocs = GridLocations(7,0,7,1)
    txtlbl = ["-3","-2","-1","0","+1","+2","+3"]
    for idx, obj in enumerate(txtlocs):
        Add(txt(txtlbl[idx], obj))
        if idx == 0:
            Add(txt("%",obj,-3))
            Add(txt("%",obj,-3))

show_object(zzz.part)
```
[image]

Jojain, 2023-03-02:
2 Is not a bug. `Location` is mutable and you are modifying one of it's attribute. It is the same as : 
```py
def editlist(a_list):
    a_list.append(2)

mylist =[1,2,3]
list_of_list = [mylist] # list_of_list = 1,2,3
editlist(mylist)
print(list_of_list) # 1,2,3,2
```

jdegenstein, 2023-03-02:
Yeah, good point, I agree.

gumyr, 2023-03-03:
I believe 1) is caused by your use of global locations instead of local locations.  If we display one of the locations:

```py
    txtlocs = GridLocations(7, 0, 7, 1)
    print(list(txtlocs)[0])
```
```
Location: (position=(-21.00, 0.00, 2.00), orientation=(-0.00, 0.00, -0.00))
```
while if we use local locations:
```py
    txtlocs = GridLocations(7, 0, 7, 1).local_locations
    print(list(txtlocs)[0])
```
```
Location: (position=(-21.00, 0.00, 0.00), orientation=(-0.00, 0.00, -0.00))
```
As these locations are being used to place the text, the global locations are lifting the text off the Plane.XY (2) in the function and then off the elevated plane in the mainline code (2+2).

This brings up the question of if a user should be allowed to add anything to a sketch that is not on the plane of the sketch. If in this case the text was placed on Plane.XY in the function it probably would have worked as expected. I'm leaning towards forcing objects to align to the sketch plane - what do you think?

jdegenstein, 2023-03-03:
Thanks for the response, that does make sense in retrospect. Would it be possible to make these equivalent?:
```py
    txtlocs = GridLocations(7, 0, 7, 1)
    txtlocs = GridLocations(7, 0, 7, 1).local_locations
```
Maybe I am oversimplifying this, but I expected the locations context to be "contained" to the variable when I assigned it, versus the typical use -- `with Locations():` where the locations context is "contained" in the indented block.

Regarding your question -- yes that sounds reasonable to me, but I am wondering if there should be a warning sent to log?

gumyr, 2023-03-04:
I don't think `GridLocations` could work that way, it's exactly the same code within a `with` block or not.  However, you might be raising a larger issue that we should consider.  Take a look at my simplified version of your code:
```py
with BuildPart(Plane.XY.offset(z_height)) as yyy:
    with BuildSketch():
        with Locations((-3.5, 0)):
            txtlocs = GridLocations(7, 0, 8, 1).local_locations
        txtlbl = ["%", "-3", "-2", "-1", "0", "+1", "+2", "+3"]
        for loc, txtstr in zip(txtlocs, txtlbl):
            with Locations(loc):
                Text(
                    txtstr,
                    font_size=3,
                    font_style=FontStyle.BOLD,
                    align=(Align.CENTER, Align.CENTER),
                )
    Extrude(amount=1)
```
[image]

The `with GridLocations()` construct works well when all of the objects being located are the same, but what if - as in this case - they aren't?  The user has to use the `zip` technique or something more complex to achieve their goals.  Maybe there should be a dedicated way to handle this fairly frequent problem?

gumyr, 2023-03-05:
I've got a potential "fix" for users adding to a sketch with a vertical offset.  Here are the logs from the above test now:
[image]

Note that the first two lines are `INFO: Face realigned to Sketch's Plane` which is telling the user that a Face was offset and has been fixed.  The `INFO: 1.9999999` shows the text at the correct elevation.

I'll need to do more testing but if this works out it won't matter if a user uses `GridLocations()` or `GridLocations().local_locations` they both should produce the same results.

closed 2023-03-06 by gumyr