gumyr/build123d #158: Should Workplanes be kept?

gumyr, 2023-03-02:
The original method of specifying a workplane for the builders was to use the `Workplanes` context. Subsequently, workplane(s) were added as input parameters to the builders which removed the need for use of the `Workplanes` context for most use cases.  Unfortunately, now there are two ways to create workplanes for the builders which results in bugs and confusion to users as to which construct to use.

What does the community think about elimination of `Workplanes` entirely?  Note that this would only remove the user's ability to directly generate a `Workplanes` context, internally this mechanism would still be used as done currently. As such, the user would still be able to pass one or more workplanes to the builders during initialization.

Jojain, 2023-03-02:
I definitly think it should be removed. I also think you should remove it internally. If we consider that you removed it for the user only then :
```py

with BuildPart() as p:
    box = Box(10, 10, 10)
    faces = box.faces()
    with BuildSketch(*faces):  # BuildSketch creates a Rect on each face.
        Rectangle(2, 5)
    Extrude(amount=10)  # Extrude extrudes normal to each face. What if I want the same orientation?
```
If we consider that it's no more allowed to work on multiple workplanes :

```py
#Does the exact same things, but simplifies a lot the internals
with BuildPart() as p:
    box = Box(10, 10, 10)
    faces = box.faces()
    for face in faces:
        with BuildSketch(face):  
            Rectangle(2, 5)
        Extrude(amount=10) # We could do Extrude(amount=10 dir = face.plane.Az) 
                           # To explicitly pass the direction (allow more control)
```

Am i missing something ? Is there something that multiple workplanes really provides over not ?

jdegenstein, 2023-03-02:
I think it was necessary before Builder contexts could be nested, but now that is not a limitation. You can even nest the same type of builder like this now:
```py
with BuildPart() as p:
    Box(3,3,3)
    with BuildPart(p.faces()[-1]):
        Box(1,1,1,align=(Align.MIN,Align.MIN,Align.MIN))
```
I agree with Jojain that `Workplanes` should be eliminated.

gumyr, 2023-03-02:
WorkplaneList (the internal version of Workplanes) was originally created to maintain orthogonality of the interfaces.  One can use multiple Locations at a time: `with Locations(a,b):` without having to use a for loop so by extension one should be able to work on multiple planes in the same way: `with Workplanes(a,b):` or `with BuildSketch(a,b):`.

Eliminating external `Workplanes` will help simplify the API though, both from an internal code point of view but more importantly from a user's point of view.

gumyr, 2023-04-10:
Workplanes have been eliminated.

closed 2023-04-10 by gumyr