gumyr/build123d #167: Scale in 2 dimensions changes a Rectangle from PLANE to BSPLINE

bernhard-42, 2023-03-09:
```python
with bd.BuildSketch() as s:
    bd.Rectangle(1,2)
    bd.Scale(by=(2,1,0))
```

returns

```text
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In [3], line 5
      3 with bd.BuildSketch() as s:
      4     bd.Rectangle(1,2)
----> 5     bd.Scale(by=(2,1,0))

File ~/Development/build123d/src/build123d/build_generic.py:520, in Scale.__init__(self, by, mode, *objects)
    515         new_object = obj_at_origin.transform_geometry(scale_matrix).locate(
    516             current_location
    517         )
    518     new_objects.append(new_object)
--> 520 context._add_to_context(*new_objects, mode=mode)
    522 super().__init__(Compound.make_compound(new_objects).wrapped)

File ~/Development/build123d/src/build123d/build_sketch.py:140, in BuildSketch._add_to_context(self, mode, *objects)
    138         obj.position = obj.position - Vector(0, 0, obj.position.Z)
    139     if isinstance(obj, Face) and not obj.is_coplanar(Plane.XY):
--> 140         raise ValueError("Face not coplanar with sketch")
    142 pre_vertices = (
    143     set()
    144     if self.sketch_local is None
    145     else set(self.sketch_local.vertices())
    146 )
    147 pre_edges = (
    148     set() if self.sketch_local is None else set(self.sketch_local.edges())
    149 )

ValueError: Face not coplanar with sketch
```

If you debug it, you see that it is coplanar, however a BSPLINE instead of a PLANE.

I think it is the same problem as https://github.com/gumyr/build123d/issues/80

gumyr, 2023-03-10:
Thanks! It's a pain that `transform_geometry` changes the object the way it does.  I'll have to change `Face.is_coplanar()` to not use the geometry.

gumyr, 2023-03-11:
Fixed with 8b3141bcee7f1cae283a563363e96e4773941b29

closed 2023-03-11 by gumyr