gumyr/build123d #174: Improved Selectors

gumyr, 2023-03-13:
When boolean operations combine Shapes, sub Shapes such as Edges can be replaced with modified versions of themselves - for example multiple shorter Edges replacing a single longer Edge.  The current `Select.LAST` algorithm will identify the shorter Edges as being new which is often not what is desired.  Discord's diz identified a way to improve the situation.  Here is an example:

```py
def join_with_fillet(a, b, r=1):
    result=a+b

    arg = TopTools_ListOfShape()
    for obj in result.edges():
        arg.Append(obj.wrapped)

    tool = TopTools_ListOfShape()
    for obj in a.edges():
        tool.Append(obj.wrapped)
    for obj in b.edges():
        tool.Append(obj.wrapped)

    operation = BRepAlgoAPI_Cut()

    operation.SetArguments(arg)
    operation.SetTools(tool)

    operation.SetRunParallel(True)
    operation.Build()
    s=Shape.cast(operation.Shape())

    return fillet(result, s.edges(), r)
```
which can be used to easily generate complex fillets like show here:
[image]

To generalize this functionality it should exist as a selector such that users can easily find the Shapes of interest.

The following should be considered:
- Should this be enhanced functionality of `Select.LAST` or should a new type of selector be created?
- The performance of doing boolean operations on all of the sub Shapes for every combination operation is likely too high to have this functionality operate always, so what needs to be stored such that this operation can be done on demand?
- Does this capability work for `Solid`, `Face`, and `Vertex` in addition to `Edge` shown here?
- To improve performance, only the sub Shapes identified by the current `Select.LAST` algorithm needs to be considered with the new functionality.

Dmytry, 2023-03-13:
Hi,

Here's my full code for the example:
```python
from alg123d import *

from cq_vscode import show, show_object, set_port
import math

def join_with_fillet(a, b, r=1):
    result=a+b

    arg = TopTools_ListOfShape()
    for obj in result.edges():
        arg.Append(obj.wrapped)

    tool = TopTools_ListOfShape()
    for obj in a.edges():
        tool.Append(obj.wrapped)
    for obj in b.edges():
        tool.Append(obj.wrapped)

    operation = BRepAlgoAPI_Cut()

    operation.SetArguments(arg)
    operation.SetTools(tool)

    operation.SetRunParallel(True)
    operation.Build()
    s=Shape.cast(operation.Shape())

    return fillet(result, s.edges(), r)

a=join_with_fillet(Box(20,5,10), Cylinder(5,10))
a=join_with_fillet(a, Sphere(2)@Pos(0,0,5))
a=join_with_fillet(a, Box(2,20,5))
#a=join_with_fillet(a, Sphere(4)@Pos(10,0,5), 2)
#a=join_with_fillet(a, Torus(8,0.5), 0.5)

show(a)
```

Regarding the implementation, a new (or modified LAST) that computes it on demand seems appropriate. 

Other thing that could be done, is selecting things by intersecting them with a solid or a face. E.g. the user may want to select all edges that are within an invisible cylinder, then bevel them. That seems to fit alg123d better though. I'm thinking something like "ab.edges() - a.edges() - b.edges()" or "a.edges() & Cylinder(...) " or the like.

Dmytry, 2023-03-13:
Having thought about it some more, I think it needs to be a new selector rather than Select.LAST, because LAST name would imply that the edges of the shape that was added to the build, should be included. 

For Select.LAST, it may make sense to work almost like the above but without b.edges() being added to the tool.

gumyr, 2023-03-14:
I tend to agree with you, probably need a new selector type. Have you tried the same technique with Faces or the other Shape classes? Being able to do: `faces(Select.CHANGED)` or `vertices(Select.CHANGED)` would be great.

Dmytry, 2023-03-14:
Vertices should work, need to try.

With faces, I think it would work but in that particular example the face list would be always empty because each face of a+b would be either somewhere on a or somewhere on b. 

Could still be useful if done without "for obj in b.edges(): tool.Append(obj.wrapped)".

Nomenclature is difficult, there's the case of (a+b).edges()-a.edges() , which is the edges that got added to the builder, and there's the case of (a+b).edges()-a.edges()-b.edges()  , which is the "truly new" edges that were created by the builder.

gumyr, 2023-08-20:
A Select.NEW selector was added with this commit: 7746f5c

closed 2023-08-20 by gumyr