gumyr/build123d #180: Algebra issue with @ operator

Dmytry, 2023-03-15:
The behavior of
```python
(Box(1,1,1) @ Pos(10,0,0) ) @ Pos(0,0,0) 
```
may be very unexpected: the Box is relocated to 0,0,0 by the second @ operator. 

Particularly confusing when it happens to an object returned from a function. A change like 

I have a suggestion regarding transformation operator. box @ Pos(...) * Rot(...)  makes it look like the box is first moved then rotated, because Pos appears closer to the box. Like row-vector convention.

I propose defining multiply operation between locations and objects, so it becomes Pos(...) * Rot(...) * obj , conforming to the familiar column vector convention that corresponds to nesting of coordinate frames within coordinate frames, culminating in an object placed in the coordinate frame. 

A multiplication between a location and an object would place the object at that location.

Dmytry, 2023-03-15:
When I originally wrote the issue I thought it was a bug, that is expected behavior.

Here's a better example:
```python
from build123d import *

from cq_vscode import show

def foo():
    result=Box(10,10,10) @ Pos(20,0,0)
    result-=Cylinder(2,50)
    return result

def new_foo():
    result=Box(10,10,10) @ Pos(20,0,0)
    # Unnecessary hole commented out
    #result-=Cylinder(2,50)
    return result

show(foo()@Pos(1,0,0), new_foo()@Pos(1,0,0))
```

My thinking is that users are likely to be caught by surprise when e.g. iterating a prototype and adding or removing removing an operation (e.g. an user decides that they don't need a fillet, or conversely, the user gets a correct end result that they print, then they decide they needed a chamfer to make it easier to get a bearing in, or something)

jdegenstein, 2023-03-15:
I think this probably IS a bug actually. Here is my version showing some oddities:
```py
def foo2():
    result=Box(10,10,10) @ Pos(20,0,0)
    result-=Cylinder(2,50) 
    return result

def new_foo():
    result=Box(10,10,10) @ Pos(20,0,0)
    # Unnecessary hole
    #result-=Cylinder(2,50)
    return result

f = foo2() @ Pos(0,20,0)
log(f"{f.location=}") #f.location=(p=(0.00, 20.00, 0.00), o=(-0.00, 0.00, -0.00))

g = new_foo() @ Pos(0,20,0)
log(f"{g.location=}") #g.location=(p=(0.00, 20.00, 0.00), o=(-0.00, 0.00, -0.00))

show_object(f,options=rand_color()) #green
show_object(g,options=rand_color()) #purple
```
[image]
The screenshot shows them in different positions, but their locations printed out are the same. Something seems wrong to me.

Dmytry, 2023-03-16:
I think it is not as much a but as just a case of general harmfulness of "location overwrite" operation in the context of an algebra.

a - b in alg123d is almost equivalent to Python operator "-" for two sets (where the sets are sets of points within shapes). For example {1,2,3} - {4} is {1,2,3} . In alg123d it would be unreasonable to expect that when a and b are disjoint, a - b is exactly identical to a, but it would be reasonable to expect that with regards to commonly used and canonical operations, a-b acts equivalently to a if a and b are disjoint sets.

gumyr, 2023-03-16:
This issue is not related to absolute or relative re-positioning operations.  Look at these greatly simplified examples:
**absolute**
```py
result_solid = Solid.make_box(10, 10, 10).locate(Location((20, 0, 0)))
print(f"{result_solid.location=}")

result_solid = result_solid.cut(Solid.make_cylinder(2, 50))
print(f"{result_solid.location=}")
```
```
result_solid.location=(p=(20.00, 0.00, 0.00), o=(-0.00, 0.00, -0.00))
result_solid.location=(p=(0.00, 0.00, 0.00), o=(-0.00, 0.00, -0.00))
```
and **relative**
```py
result_solid = Solid.make_box(10, 10, 10).move(Location((20, 0, 0)))
print(f"{result_solid.location=}")

result_solid = result_solid.cut(Solid.make_cylinder(2, 50))
print(f"{result_solid.location=}")
```
```
result_solid.location=(p=(20.00, 0.00, 0.00), o=(-0.00, 0.00, -0.00))
result_solid.location=(p=(0.00, 0.00, 0.00), o=(-0.00, 0.00, -0.00))
```
The problem is present in both.

Displaying the location of the bounding box is helpful to understand what is happening:
```py
result_solid = Solid.make_box(10, 10, 10).move(Location((20, 0, 0)))
print(f"{result_solid.location=}")
print(f"{result_solid.bounding_box().min=},{result_solid.bounding_box().max=}")

result_solid = result_solid.cut(Solid.make_cylinder(2, 50))
print(f"{result_solid.location=}")
print(f"{result_solid.bounding_box().min=},{result_solid.bounding_box().max=}")
```
```
result_solid.location=(p=(20.00, 0.00, 0.00), o=(-0.00, 0.00, -0.00))
result_solid.bounding_box().min=Vector: (20.0, 0.0, 0.0),result_solid.bounding_box().max=Vector: (30.0, 10.0, 10.0)
result_solid.location=(p=(0.00, 0.00, 0.00), o=(-0.00, 0.00, -0.00))
result_solid.bounding_box().min=Vector: (20.0, 0.0, 0.0),result_solid.bounding_box().max=Vector: (30.0, 10.0, 10.0)
```
Here we can see that even though the `location` is reset back to the origin after the `cut` operation, the position of the object in space has not changed as illustrated by the bounding boxes being equal.  This is caused by OCCT recreating the box in the `cut` operation with vertices in different positions. Normally this wouldn't be an issue but as the algebra api exposes the `location` to the user in a very direct way this could be a serious problem which needs further study.

Dmytry, 2023-03-17:
Also case in point, OpenSCAD does not seem to expose any location-overwriting whatsoever.

I think location overriding moves are most relevant to GUI code. For example a mouse drag may read the location, then set up the screen space offset vs the mouse pointer, then overwrite location as its being dragged.

Or a constraint solver may read the location, use it as a hint, then write it.

Or a button could be provided that resets the object location, to help the user start their work on that object's placement anew.

I don't think it is very relevant to codecad context. If the transform being overwritten is an identity transform, the overwrite is equivalent to multiplication, if the transform being overwritten is not an identity transform, that is more likely to be unintentional.

gumyr, 2023-04-12:
Algebra no longer uses the @ operator to locate objects.  This page describes how location is done: https://build123d.readthedocs.io/en/latest/location_arithmetic.html

closed 2023-04-12 by gumyr