gumyr/build123d #198: Workplanes aren't impacted by Locations

gumyr, 2023-04-03:
```py
with BuildPart() as plateLeft:
    with Locations((0,0,10)):
        with BuildSketch() as sketch:
            make_face(maggie.silhouetteLeft())
            add(maggie.getSwitchCutouts(True), mode=Mode.SUBTRACT)
            add(maggie.getMagnetHoles(True), mode=Mode.SUBTRACT)
    extrude(amount=1.5)
```
This is not moving the plate to 10 in the Z dir - not even if extrude is indented by 1.

jdegenstein, 2023-04-03:
```py
from build123d import *

def foo():
    with BuildSketch() as s:
        Circle(1)
    return s.sketch

with BuildPart() as mainp:
    with Locations((0,10,10)):
        with BuildPart() as cp:
            with BuildSketch() as mains:
                Rectangle(10,10)
                add(foo(), mode=Mode.SUBTRACT)
            extrude(amount=1.5)

show_object(mainp.part,options=rand_color())
show_object(cp.part,options=rand_color())
```
[image]
I expected this to work, but it didn't (no z-translation on either `cp` or `mainp` -- they are coincident). This alternate version from bernhard (re-using the same `foo()`) works fine:
```py
with BuildPart() as cp:
    with BuildSketch() as mains:
        Rectangle(10,10)
        add(foo(), mode=Mode.SUBTRACT)
    extrude(amount=1.5)

with BuildPart() as mainp:
    with Locations((0,10,10)):
        add(cp.part)
        
show_object(mainp.part,options=rand_color())
show_object(cp.part,options=rand_color())
```
[image]

gumyr, 2023-04-20:
This issue reflects a design choice.  The even simpler example is this:

```py
from build123d import *
from build123d import LocationList

with Locations((1, 1, 1)):
    print(f"{LocationList._get_context().locations=}")
    with BuildPart() as wp_test:
        print(f"{wp_test.workplanes=}")
        print(f"{LocationList._get_context().locations=}")
```
```
LocationList._get_context().locations=[(p=(1.00, 1.00, 1.00), o=(-0.00, 0.00, -0.00))]
wp_test.workplanes=[Plane(o=(0.00, 0.00, 0.00), x=(1.00, 0.00, 0.00), z=(0.00, 0.00, 1.00))]
LocationList._get_context().locations=[(p=(-0.00, 0.00, -0.00), o=(-0.00, 0.00, -0.00))]
```

The design assumption that drives this is that locations apply to a specific workplane(s) and creating a new workplane creates a new entry in the location context at the origin.

If locations did impact workplanes then:
```py
with Locations((0, 0, 0), (90, 0, 0)):
    with BuildPart(Plane.XY):
        Box(1, 1, 1)
```
would not be creating the Box on `Plane.XY` but actually `Plane.YZ` with no ability actually set the workplane to `Plane.XY` other than starting with `Plane.YZ` which would be very confusing.

However, one can create workplanes from locations so this works:
```py
with BuildPart() as cp:
    with BuildSketch(Location((0, 10, 10))) as mains:
        Rectangle(10, 10)
        add(foo(), mode=Mode.SUBTRACT)
    extrude(amount=1.5)
```
[image]

I'm hesitant to change this behaviour but maybe you can convince me.

closed 2023-06-18 by gumyr