gumyr/build123d #200: Documentation: return type of Compound.do_children_intersect is wrong

barnabywalters, 2023-04-03:
The documentation says that [Compound.do_children_intersect()](https://build123d.readthedocs.io/en/latest/direct_api_reference.html#topology.Compound.do_children_intersect) returns `Bool`, but in fact it returns a tuple of `(Bool, (tuple of intersecting parts), volume of intersection)`, e.g.

`(True, (Part at 0x13333a250, label(), #children(0), Part at 0x1322857c0, label(), #children(0)), 9094.390125741316)`

barnabywalters, 2023-04-04:
We could consider returning `(False, (), 0.0)` if no intersection is found, so that code consuming the other items doesn’t have to add an additional conditional branch for handling them being `None`

gumyr, 2023-04-11:
Fixed as discussed with the improved return code - thanks for pointing this out.  46d87eafa78780f9ab7722dcfa317c8220f071cb

closed 2023-04-11 by gumyr