gumyr/build123d #223: Allow extruding until a Face in the opposite direction

JonnyHaystack, 2023-04-24:
As the title says, it would be nice to have a more convenient way to change the direction of an extrude until face operation. E.g. if I have `extrude(until=Until.LAST, mode=Mode.SUBTRACT)` and I want it to extrude to a Face in the opposite direction, I would like to be able to do something like `extrude(until=-Until.LAST, mode=Mode.SUBTRACT)` as the default extrude direction tends to be in the wrong direction when you want to cut through a solid.

gumyr, 2023-05-15:
I've extended the Until Enum with PREVIOUS and FIRST (previous just NEXT and LAST) and added functionality to the `extrude` operation to enable these new options (76dd8cd801e58a17fae7f5c5980ac7864fe40d90).  However, I'm wondering if the better solution is to change the `Solid.extrude_until` method - seems like the right way to go on second thought.

gumyr, 2023-05-16:
Moved to the `Solid.extrude_until` method with commit 70e0c14da8a5f400437ca339ecd9c5d8a24b7cfe

gumyr, 2023-05-16:
Added new ENUMs to cheat sheet f89fbfce79d4a9a8e2a465a20e66e3cde200f054

closed 2023-05-16 by gumyr