gumyr/build123d #225: Wedge with 0 Z size fails

fnordpig, 2023-04-25:
When trying to make a wedge without a "flat top" this fails:

```python
Wedge(2, 1, 0, 0, 0, 2, 5)
```

with this error:

```
  File "/opt/homebrew/Caskroom/miniconda/base/envs/cq-b123d/lib/python3.10/site-packages/build123d/objects_part.py", line 549, in __init__
    solid = Solid.make_wedge(xsize, ysize, zsize, xmin, zmin, xmax, zmax)
  File "/opt/homebrew/Caskroom/miniconda/base/envs/cq-b123d/lib/python3.10/site-packages/build123d/topology.py", line 4964, in make_wedge
    ).Solid()
OCP.Standard.Standard_Failure  
```

The work around is to do something like:

```
Wedge(2, 1, 0.0000001, 0, 2, 5)
```

Which makes a wedge with an epsilon length flat top.

jdegenstein, 2023-04-25:
looks like the second wedge was missing the 7th parameter, here it is corrected
```py
Wedge(2, 1, 0.0000001, 0, 0, 2, 5)
```

gumyr, 2023-06-19:
None of the first three parameters: xsize, ysize, zsize can be zero with the OCP BRepPrimAPI_MakeWedge.  I find this class too complex - 7 dimensional parameters seems like too much.  Should it remain when it's easy to create a sketch of whatever shape and extrude it?

gumyr, 2023-08-18:
Added a check and error measure for the first three values being > 0.

closed 2023-08-18 by gumyr