gumyr/build123d #228: [Enhancement] Relative Line

jdegenstein, 2023-04-27:
instead of needing to invoke a position twice for a relative Line movement, there should be a dedicated method:
```py
        with BuildLine() as l:
            l1 = Line((0,25),(250/2-15,25))
            l2 = Line(l1@1,l1@1+(0,30/2))
            l2 = RelLine(l1@1,(0,30/2)) #automatically invokes the second parameter as relative to the first
```

fischman, 2023-11-05:
A related but separate proposal (originally in [discord](https://discord.com/channels/964330484911972403/1074840524181217452/1170771851325231297)): `BaseLineObject`s could use the `@1` of the immediately-preceding `BaseLineObject` for their `start_point` (or equivalent) instead of requiring the user to explicitly state this.

snoyer, 2023-11-06:
related: #191

jdegenstein, 2024-08-23:
@fischman Following up on your comment above, I personally do not like the proposal to automatically use `@1` as there are often times when I am creating a series of segments in the clockwise direction, and then switch to counterclockwise because of what is known about the design. Here is a simple example showing a situation in which the angle/length of a line segment is not specified:

[image]

fischman, 2024-08-23:
@jdegenstein any reason not to use @1 as default, but let user specify an override if needed?

jdegenstein, 2024-08-23:
I would prefer to just keep it simple and let the user provide the Vector representing the starting point which is the way most  current BaseLineObject-derived classes currently work. End users can always write custom helpers that override this behavior if needed.

unprolix, 2025-11-09:
here's what i'm using for this for now:

```python
class BuildChainedLine(BuildLine):
    """BuildLine that tracks the last endpoint as 'last'."""
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self.last = None
        self.X = None
        self.Y = None
    
    def lineTo(self, point):
        return Line(self.last, point)

    def continueWithThreePointArc(self, p1, p2):
        return ThreePointArc(self.last, p1, p2)

    def _add_to_context(self, obj, mode=None):
        result = super()._add_to_context(obj, mode=mode)
        self.last = obj @ 1
        self.X = self.last.X
        self.Y = self.last.Y
        return result```