gumyr/build123d #239: project misses an Edge

gumyr, 2023-05-16:
```py
with BuildPart() as loaf_l:
    with BuildSketch(Plane.YZ) as profile:
        Rectangle(10, 15)
        fillet(profile.vertices().group_by(Axis.Y)[-1], 2)
    extrude(amount=15, both=True)
    ref_s_pnts = loaf_l.vertices().group_by(Axis.X)[-1].group_by(Axis.Z)[0]
    origin = (ref_s_pnts[0].to_vector() + ref_s_pnts[1].to_vector()) / 2
    x_dir = ref_s_pnts.sort_by(Axis.Y)[-1] - ref_s_pnts.sort_by(Axis.Y)[0]
    workplane_l = project_workplane(origin, x_dir, (1, 0, 1), 30)
    with BuildLine(workplane_l) as projection_l:
        e = loaf_l.part.faces().sort_by(Axis.X)[-1].edges()
        p = project(e)
        Spline((-4, 0), (0, -4), (4, 0))
```
[image]

Even though all the Edges on the end are selected (shown in red), the bottom one doesn't get projected.

gumyr, 2023-06-19:
The problem is within `BRepProj_Projection`.  The missing edge produces no result at all while I can't determine any significant way that it's different from the other edges in this example.

bernhard-42, 2023-08-18:
Looked into it and it is an interesting bug:
- When run without debugger, the last line is missing
- When run in debugger it is there
So I thought it has to do with Python <=> C++ comms or maybe memory issues.

I changed 
    `target = Face.make_rect(1e9, 1e9, plane=workplane)` 
to 
    `target = Face.make_rect(1e6, 1e6, plane=workplane)` 
in `operations_generic.py:703` and, voila, it works for me.

So, what so ever happens in the kernel, the `target` seems to be to large.

Is there a chance to chage the algorithm to not depend on a "infinitely" large plane?

gumyr, 2023-08-18:
Very interesting. Thinking more about the target size,: given that the target is normal to the direction of projection, for a parallel projection the target can be no larger than the largest dimension of the object's bounding box. Double that for object alignment and it should always fit, no?

bernhard-42, 2023-08-18:
In theory one could project edges onto a plane in any direction that is not parallel to the plane (so not only the normal). Then the result could be much larger than the input.
But as this routine only supports projection in normal direction, I'd say double the largest bounding box dimension or the bounding box sphere diameter should be ok.

gumyr, 2023-08-19:
Thanks for determining the cause of this problem @bernhard-42 - changing the target to 3 times (just to be safe) the maximum object dimension fixes the problem.
[image]

closed 2023-08-19 by gumyr