gumyr/build123d #248: Euler Rotation Order as Parameter for Rotation

ZacharyKF, 2023-06-05:
Right now rotations are limited to the XYZ rotation order through the constructor. While a workaround exists where the user can explicitly apply rotations separately in the desired order, a more succinct solution would be to add an "ordering" parameter to the `Rotation` constructor. This would allow for various domain specific design paradigms. Something like:
`Rotation((a, b, c), ordering=ZYZ)`
Going further, having the ability to specify the default `ordering` would make build123d more friendly to disciplines like robotics.

jdegenstein, 2023-10-09:
should `Rotation((a, b, c), ordering=ZYZ)` use an enum instead, e.g. `Rotation((a, b, c), ordering=RotationOrder.ZYZ)` ?

naming it `AxisOrder.ZYZ` instead might be a bit more general for future re-use.

Do we plan to support only intrinsic rotations, or all at this page including extrinsic?: https://dev.opencascade.org/doc/refman/html/gp___euler_sequence_8hxx.html

There are 12 intrinsic rotation orders available at the link above (and 12 extrinsic).

ZacharyKF, 2023-10-09:
An enum definitely seems like the best solution. Having access to all possible rotations seems like a smart idea to improve accessibility, not sure about the effort required to support this though.

jdegenstein, 2023-10-09:
The effort is quite small to support extrinsic too. Also, it appears that OpenSCAD uses extrinsic rotations, so I think having extrinsic available here is a good idea for newcomers from that community.

closed 2023-12-06 by gumyr