gumyr/build123d #252: Extrusion with taper can flip direction.

jrmobley, 2023-06-10:
When the `section` face for the extrusion is constructed in a particular way, and a non-zero `taper` is specified, `Solid.extrude_linear` will extrude in the opposite direction.  I apologize for the somewhat lengthy repro case, but I was unable to simplify it further and still trigger the issue.

```py
# Build a box.
with BuildPart() as buildy:
    Box(10, 10, 5)
box = buildy.part

# Build a plate inside the box.
box_top_face = box.faces().filter_by(Axis.Z).sort_by(Axis.Z)[-1]
box_top_plane = Plane(box_top_face, x_dir=Vector(1,0,0))
plate_top_plane = box_top_plane.offset(-3)
plate_points = [(-4, 4), (4, 4), (4, -4), (-4, -4)]
with BuildPart() as buildy:
    with BuildSketch(plate_top_plane):
        Polygon(*plate_points, align=None)
    extrude(amount=-1)
plate = buildy.part
show_object(plate, name='plate', options={'color':'green'})

# Cut a pocket in the box to fit the plate.
pocket_taper = -5
plate_bottom_face = plate.faces().filter_by(Axis.Z).sort_by(Axis.Z)[0]
pocket_bottom_face = -plate_bottom_face
pocket_bottom_plane = Plane(pocket_bottom_face, x_dir=Vector(1,0,0))
with BuildPart() as buildy:
    add(box)
    with BuildSketch(pocket_bottom_plane):
        if False:
            # creating the section from points produces expected results.
            Polygon(*plate_points, align=None)
        else:
            # creating section from existing faces results in a flipped extrude.
            section = pocket_bottom_plane.to_local_coords(pocket_bottom_face)
            print("plane normal", pocket_bottom_plane.z_dir)
            print("section normal", section.normal_at())
            add(section)
    extrude(amount=4, taper=pocket_taper, mode=Mode.SUBTRACT)
box = buildy.part
show_object(box, name='box')
```

gumyr, 2023-07-02:
Narrowed the problem down:
```py
from build123d import *
from ocp_vscode import *

f_up = Face.make_rect(8, 8)
f_down = -f_up

s_up = Solid.extrude_linear(f_up, normal=(0, 0, 1), taper=-4)
s_down = Solid.extrude_linear(f_down, normal=(0, 0, 1), taper=-4)
show_all()
```
[image]

The face normal is impacting the direction of extrusion even though the `normal` parameter is the same in both cases.

gumyr, 2023-07-02:
Fixed:
[image]

jrmobley, 2023-07-03:
Ah yes... I think I recall what happened here.  Very early in my b3d experience, I reversed the winding of my sketch in order to get my extrusion to go in the direction I wanted.  Now I know enough to see that is backwards and I must have been encountering this bug back then, and I just didn't know enough to see it.

And this now has taught me that you *can* produce a backward facing 2D sketch by controlling the winding order in `BuildLine`.  I need to go revise my advice in the 2D Exports discussion.

jrmobley, 2023-07-03:
This should probably be a separate issue, but your test case already illustrates that a negative taper introduces a fillet, which is definitely not what I would expect.

gumyr, 2023-07-03:
> This should probably be a separate issue, but your test case already illustrates that a negative taper introduces a fillet, which is definitely not what I would expect.

This is a consequence of using the LocOpe_DPrism to create the extrusion; there is no option to change the fillet behavior.  Maybe a better solution is to not use LocOpe_DPrism but BRepOffsetAPI_ThruSections to create a loft.

gumyr, 2023-07-05:
Fixed with commit c83b52e998f79c5363afb24cd3f3a24a9057febf

closed 2023-07-05 by gumyr