gumyr/build123d #259: Add examples for import_svg and import_svg_as_buildline_code

gumyr, 2023-06-21:
```py
# https://en.wikipedia.org/wiki/Spades_(suit)#/media/File:SuitSpades.svg

with BuildPart() as example:
    with BuildSketch():
        with BuildLine():
            add(import_svg("SuitSpades.svg"))
        make_face()
    extrude(amount=10)

print(import_svg_as_buildline_code("SuitSpades.svg"))

show(example)
```

zhuewizz, 2023-07-24:
i use your SuitSpades.svg,export_step("box5.step") is sucess
but i use my svg export_step is error
i use :
from svgpathtools import svg2paths
paths, attributes= svg2paths('svg/SuitSpades.svg')
redpath = paths[0]
attr = attributes[0]
print(redpath)
output is:
Path(CubicBezier(start=(420.37498+349.55102j), control1=(360.40052+390.08152j), control2=(320.37505+467.57634j), end=(320.375+514.71623j)),
     CubicBezier(start=(320.375+514.71623j), control1=(320.375+539.21657j), control2=(340.29885+560.81235j), end=(366.92151+560.81235j)),
     CubicBezier(start=(366.92151+560.81235j), control1=(388.48053+560.81235j), control2=(406.86151+541.23743j), end=(406.86146+514.26579j)),
     CubicBezier(start=(406.86146+514.26579j), control1=(406.86146+510.20054j), control2=(408.9958+508.40994j), end=(410.76536+508.40994j)),
     CubicBezier(start=(410.76536+508.40994j), control1=(413.08604+508.40994j), control2=(415.42002+511.71285j), end=(415.42002+518.01954j)),
     CubicBezier(start=(415.42002+518.01954j), control1=(415.42002+533.65083j), control2=(407.57185+573.31705j), end=(396.95155+597.44898j)),
     CubicBezier(start=(396.95155+597.44898j), control1=(403.08128+595.63666j), control2=(411.25603+593.8454j), end=(420.37498+593.8454j)),
     CubicBezier(start=(420.37498+593.8454j), control1=(429.49402+593.8454j), control2=(437.81882+595.63666j), end=(443.94856+597.44898j)),
     CubicBezier(start=(443.94856+597.44898j), control1=(433.32825+573.31705j), control2=(425.48008+533.65083j), end=(425.48008+518.01954j)),
     CubicBezier(start=(425.48008+518.01954j), control1=(425.48008+511.71285j), control2=(427.81406+508.40994j), end=(430.13474+508.40994j)),
     CubicBezier(start=(430.13474+508.40994j), control1=(431.9043+508.40994j), control2=(433.88849+510.20054j), end=(433.88849+514.26579j)),
     CubicBezier(start=(433.88849+514.26579j), control1=(433.88854+541.23743j), control2=(452.41957+560.81235j), end=(473.97859+560.81235j)),
     CubicBezier(start=(473.97859+560.81235j), control1=(500.60125+560.81235j), control2=(520.375+539.21657j), end=(520.375+514.71623j)),
     CubicBezier(start=(520.375+514.71623j), control1=(520.37505+467.57634j), control2=(480.34948+390.08152j), end=(420.37498+349.55102j)))

i change my 1.svg   run output ：
Path(CubicBezier(start=(97.6+142.7j), control1=(98.3+148.6j), control2=(96.6+150.39999999999998j), end=(90.8+149.79999999999998j)),
     CubicBezier(start=(90.8+149.79999999999998j), control1=(87.8+149.49999999999997j), control2=(87.5+149.1j), end=(86.5+144.49999999999997j)),
     Line(start=(86.5+144.49999999999997j), end=(85.4+139.49999999999997j)),
     Line(start=(85.4+139.49999999999997j), end=(83.60000000000001+142.19999999999996j)),
     CubicBezier(start=(83.60000000000001+142.19999999999996j), control1=(79.60000000000001+148.29999999999995j), control2=(80.2+152.59999999999997j), end=(85.80000000000001+158.29999999999995j)),
     CubicBezier(start=(85.80000000000001+158.29999999999995j), control1=(90.20000000000002+162.79999999999995j), control2=(93.00000000000001+167.89999999999995j), end=(106.00000000000001+194.49999999999994j)),
     CubicBezier(start=(106.00000000000001+194.49999999999994j), control1=(133.5+250.59999999999994j), control2=(132.10000000000002+246.89999999999995j), end=(128+254.99999999999994j)),
     CubicBezier(start=(128+254.99999999999994j), control1=(125.3+260.29999999999995j), control2=(126.3+264.99999999999994j), end=(131.1+270.49999999999994j)),
     CubicBezier(start=(131.1+270.49999999999994j), control1=(135.4+275.3999999999999j), control2=(136.4+274.99999999999994j), end=(135.6+268.59999999999997j)),
     CubicBezier(start=(135.6+268.59999999999997j), control1=(134.79999999999998+262.4j), control2=(136.7+259.99999999999994j), end=(142.6+259.99999999999994j)),
     CubicBezier(start=(142.6+259.99999999999994j), control1=(146.5+259.99999999999994j), control2=(147.6+261.29999999999995j), end=(148.5+266.69999999999993j)),
     CubicBezier(start=(148.5+266.69999999999993j), control1=(149.4+272.29999999999995j), control2=(150.5+272.79999999999995j), end=(152.6+268.79999999999995j)),
     CubicBezier(start=(152.6+268.79999999999995j), control1=(156.5+261.49999999999994j), control2=(155.2+255.69999999999996j), end=(148.4+249.29999999999995j)),
     CubicBezier(start=(148.4+249.29999999999995j), control1=(142.6+243.69999999999996j), control2=(141.5+241.79999999999995j), end=(122.2+203.29999999999995j)),
     CubicBezier(start=(122.2+203.29999999999995j), control1=(101.2+161.59999999999997j), control2=(100.5+159.79999999999995j), end=(103.6+153.79999999999995j)),
     CubicBezier(start=(103.6+153.79999999999995j), control1=(105.8+149.49999999999994j), control2=(104.69999999999999+145.09999999999997j), end=(100.39999999999999+140.89999999999995j)),
     Line(start=(100.39999999999999+140.89999999999995j), end=(96.89999999999999+137.49999999999994j)),
     Line(start=(96.89999999999999+137.49999999999994j), end=(97.6+142.69999999999993j)),
     Line(start=(97.6+142.69999999999993j), end=(97.6+142.7j)))

Can you look at it for me? Is there something wrong with my svg file format？ tks

zhuewizz, 2023-07-24:
[image]

zhuewizz, 2023-07-24:
[image]
this is  output  error info

gumyr, 2023-07-24:
I'm already got a fix for this problem on the go but there is an easy work-around.  As part of the svg import, python code is generated with the svg file name as a python identifier.  `3` is a valid file name but not a valid python identifier so the import fails.  As a work around just change the file name to `three.svg` (or any other valid python identifier) and this problem should go away.