gumyr/build123d #279: Sometimes spliting an object does not work correctly

asicdruide, 2023-07-15:
The issue was debugged by @bernhard-42.

"If you debug into the split function, it creates a cutter box, and this box is not properly positioned, see screenshot"

[image]

gumyr, 2023-07-15:
```py
y0 = 0
y1 = 365
y2 = 563
x0 = 0
x1 = 365/2
x2 = 365

points = [(x0,y0)
         ,(x0,y1)
         ,(x1,y2)
         ,(x2,y1)
         ,(x2,y0)
         ]

plane_lr = Plane.YZ.offset(x1)
plane_ud = Plane.ZX.offset(280)

with BuildPart() as all:
    with BuildSketch() as sk_frame:
        with BuildLine() as frame_outer:
            Polyline(*points , close=True)
        make_face()

    extrude(amount=2)
    
    lower       = split(all.part , bisect_by=plane_ud , keep=Keep.BOTTOM, mode=Mode.PRIVATE)
    upper       = split(all.part , bisect_by=plane_ud , keep=Keep.TOP   , mode=Mode.PRIVATE)
    lower_left  = split(lower    , bisect_by=plane_lr , keep=Keep.BOTTOM, mode=Mode.PRIVATE)
    lower_right = split(lower    , bisect_by=plane_lr , keep=Keep.TOP   , mode=Mode.PRIVATE)
    upper_left  = split(upper    , bisect_by=plane_lr , keep=Keep.BOTTOM, mode=Mode.PRIVATE)
    upper_right = split(upper    , bisect_by=plane_lr , keep=Keep.TOP   , mode=Mode.PRIVATE)

show_object(upper      ,name="upper")
show_object(lower      ,name="lower")
show_object(lower_left ,name="lower_left")
show_object(lower_right,name="lower_right")
show_object(upper_left ,name="upper_left")
show_object(upper_right,name="upper_right")
```

gumyr, 2023-07-17:
I've changed the implementation of `Shape.split` and used that to refactor the `split` operation.  The above code produces the following now:
[image]

closed 2023-07-17 by gumyr

asicdruide, 2023-07-18:
Thank you for your comprehensive support and for fixing this issue!