gumyr/build123d #280: BuildLine interaction between non-global origin and tuple addition/localization

jdegenstein, 2023-07-18:
```py
pln = Plane.XZ
pln.origin = (0,0,-60+25)
with BuildLine(pln) as l2:
    n1 = Line((-270/2,-40+5),(-270/2+30,-40+5))
    n2 = PolarLine(n1@1,70,30)
    n3 = Line(n2@1,(0,0))
    n4 = Line(n3@1,n3@1+(0,10))
print(f"{n4@0=}, {n4@1=}")
#returns n4@0=Vector: (0.0, 0.0, -35.0), n4@1=Vector: (0.0, 0.0, -60.0)
```
[image]

If I comment out `pln.origin = (0,0,-60+25)` it works as expected (line goes up instead of down)

gumyr, 2023-08-18:
[image]

The problem was with localization of vector/tuple addition when the plane origin isn't (0,0,0).

closed 2023-08-18 by gumyr

jdegenstein, 2023-08-28:
Thank you Roger!