gumyr/build123d #281: DXF export issue with ellipses

voneiden, 2023-07-23:
Ref https://github.com/CadQuery/cadquery/issues/1381 - Build123d exhibits same issue. 

MRE:

```python
import build123d as b3d
b3d_shape = b3d.import_brep("buggy.brep")
b3d_shape.export_dxf("buggy_b3d.dxf")
```

Probably tangentially relevant to #75

gumyr, 2023-07-24:
There is a new much improved SVG and DXF exporter that exists but hasn't been documented yet which will entirely replace the current `export_dxf` method.  If you'd like to try it let me know and I'll guide you through its use.

gumyr, 2023-07-24:
There is now documentation at: https://build123d.readthedocs.io/en/latest/import_export.html#d-exporters

voneiden, 2023-07-25:
Perfect, it works as expected with the new exporter, thank you very much.  :+1:

closed 2023-08-13 by gumyr