gumyr/build123d #289: pip install doesn't work

PoeticPete, 2023-07-30:
Running `python3 -m pip install git+https://github.com/gumyr/build123d` resulted in
```
ERROR: Could not find a version that satisfies the requirement cadquery-ocp~=7.7.0a0 (from build123d) (from versions: none)
```
Also tried `python3 -m pip install git+https://github.com/gumyr/build123d.git#egg=build123d`, which had the same error

PoeticPete, 2023-07-30:
Update I tried the apple silicon workaround: https://github.com/gumyr/build123d/blob/3da1615b38302febf7182b14ff87b28b19937cdf/docs/installation.rst#special-notes-on-apple-silicon-installs

Now I'm running into this error:

```
(cad_stuff) ➜  cad_stuff python
Python 3.10.12 | packaged by conda-forge | (main, Jun 23 2023, 22:41:52) [Clang 15.0.7 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from build123d import *
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/petertao/homebrew/anaconda3/envs/cad_stuff/lib/python3.10/site-packages/build123d/__init__.py", line 10, in <module>
    from build123d.importers import *
  File "/Users/petertao/homebrew/anaconda3/envs/cad_stuff/lib/python3.10/site-packages/build123d/importers.py", line 34, in <module>
    from stl.mesh import Mesh
ModuleNotFoundError: No module named 'stl'
```

PoeticPete, 2023-07-30:
Running `pip install numpy-stl` fixed this error

closed 2023-07-30 by PoeticPete

gumyr, 2023-07-30:
The `numpy-stl` problem is strange as it's listed as a dependency in the `pyproject.toml` file:
```
dependencies = [
    "cadquery-ocp ~= 7.7.0a0",
    "OCP-stubs @ git+https://github.com/CadQuery/OCP-stubs@7.7.0",
    "typing_extensions >= 4.4.0, <5",
    "numpy >= 1.24.1, <2",
    "svgpathtools >= 1.5.1, <2",
    "anytree >= 2.8.0, <3",
    "ezdxf >= 1.0.0, < 2",
    "numpy-stl >= 3.0.0, <4"
]
```
and therefore should be installed automatically.

PoeticPete, 2023-07-30:
I think it's because of the `--no-deps`. I just copy and pasted :P 

```
pip install --no-deps git+https://github.com/gumyr/build123d svgwrite svgpathtools anytree scipy
```