gumyr/build123d #301: Compound export_step doesn't keep its child's positions

longhan, 2023-08-17:
When create a compound name 'assembly' by add its children (also a compose, by using import_step), then using joins to relocates the children.

the ocp_vscode viewer showed the root compound perfectly with its children being relocated.
[image]

But when I use the assembly.export_step(path) and using Freecad to import. All the children locations is the same to the assembly before relocated.
[image]

gumyr, 2023-08-20:
This one is going to be difficult to fix as I don't know how to reproduce it.  Here is the latch and slide (from https://build123d.readthedocs.io/en/latest/joints.html#linear-joint) exported as an Compound/assembly to a step file and imported to FreeCAD:
[image]

Is there anything else you could share to help narrow down the problem?

Gigahawk, 2023-09-07:
From discussions on discord, this may have something to do with the way OCCT is used to import step files.
Below is the sample STEP file that demonstrates the issue:
[reprapdiscount_smart_controller.zip](https://github.com/gumyr/build123d/files/12552729/reprapdiscount_smart_controller.zip)

This file was generated by [this GitHub Action](https://github.com/Gigahawk/reprapdiscount_smart_controller/actions/runs/6104094055), which uses https://github.com/Gigahawk/freecad-export-action to export the KiCad design as a STEP file through KiCadStepUp

When importing this file in build123d, there is an extra set of solids placed at `Location((0, 0, 0), (0, 0, 0))` (screenshot shows cq-editor, but the same behavior is seen in ocp-vscode)
[image]

# Working examples

When this file is opened with [CAD Assistant](https://www.opencascade.com/products/cad-assistant/), it looks fine
[image]

It also looks fine in the latest version of Mayo (0.7.0) (the extra solids are named `SolidXXX` in the tree, enabling/disabling them doesn't do anything)
[image]

It also seems to be fine in FreeCAD 0.20.2 (I have not found any set of import settings that causes it to break)
[image]

# Broken examples

In the previous Mayo release (0.6.0), the extra solids are shown:
[image]

They are also shown in Fusion:
[image]

# Misc

In Mayo 0.7.0, enabling/disabling the extra solids seems to have no effect
[image]

In CAD Assistant, changing the transparency on one of the extra solids changes the transparency of the corresponding solid in the correct object:
[image]

In Mayo 0.6.0, enabling/disabling the extra solid is linked only to the extra solid:
[image]

closed 2026-05-04 by gumyr