gumyr/build123d #305: Unfold Sheet Metal Parts

gumyr, 2023-08-24:
Take a model designed in 3D representing a part made from sheet metal and unfold it back to a planar piece of sheet metal as a cutting pattern. Roughly it work works as follows:

The algorithm that I've come up with is:
1) Find all of the Faces that are continuous with each other - this separates the "edges" of the sheet metal from the surface of the sheet metal.  The attached screenshot shows some continuous Faces (pink) and a translucent non-continuous Face in the back.  Unfortunately, I found that the OCP BRep_Tool.Continuity_s tool's result vary depending on how the faces were constructed so I've got more work to do here.
2) Pick a non-planar Face identified in step 1, generate a new planar version with "the same" area (actually the area will be a little different based on the "k-factor" of the material - the physics of bending stuff). I'm going to assume that the bend is circular to make the math manageable. This new planar Face will contain the common Edge from the adjoined planar Face.
3) Sew the Faces together to create a new part with one less bend.
4) Repeat steps 2 and 3 until there are no more non-planar Faces.
5) Extrude the newly formed planar Faces back into a 3D sheet metal part. The extrusion amount is determined from the sheet metal "edges" found in step 1.

kcleung, 2025-01-07:
Some thoughts: for the unfolding function, perhaps we can have two variants:

* laser cut and CNC bend - where the SVG represents the following layers: untouched, cut, top engrave, bottom engrave, fold line
* photoetch - where the SVG have layers representing the following layers: unetched, fully-edged, top-edged and bottom-edged layers.

However, for the fold lines, how can we represent the angle and radius of each fold line in the SVG or DXF file?  This information is vital for CNC fold.

Furthermore, I understand from my experience in building from brass locomotive kits that in the case of photoetch, it is a common practice to convert fold lines into a half etch on the inner side of the fold, so that the user can easily fold the sheet correctly with minimal tools.

Can people with experience in laser cut, waterjet cut, CNC fold and photoetching share your thoughts?

JHNitiu, 2026-04-02:
Hey there!

First off, thanks for the awesome CAD library—it’s been super helpful for our team.
I noticed there’s no sheet metal support yet. Is this something you’re already working on? A few of us have a real need for it and are thinking about contributing. Would this be a useful addition, or is there a specific direction you’d prefer for contributions?

Looking forward to your thoughts!

gumyr, 2026-04-02:
@JHNitiu I'm happy to hear that build123d is useful to you. This issue is specific to flattening a sheet metal design as described in the issue description above - is that what "sheet metal support" means to you? 

No work is currently underway related to this feature as it isn't gating the 1.0.0 release. However, if you're willing to contribute towards this effort that would be greatly appreciated. I can guide you through some of the code to find adjacent faces and navigate the shell of the object if you like. If you aren't already familiar with the "k-factor" you'll need to do some reading as that described how the bend works mathematically.

There is work currently underway on the material system for build123d where a use will be able to set the `material` attribute of a shape thus providing the material information required for the k-factor calculation.

gumyr, 2026-04-02:
It would be a good idea to let me know what you plan to do before you get too far to avoid future issues.

gumyr, 2026-10-07:
Complete with PR #1434

closed 2026-10-07 by gumyr