gumyr/build123d #311: Certain Fonts/Characters Break the Lib3MF Mesher()

jdegenstein, 2023-09-01:
When using the Lib3MF Mesher() it was noticed that the Lib3MF library sometimes returns that the mesh is invalid. After further investigation I noticed that certain combinations of fonts/characters seem to be the "root cause" (as was the case in the originally found example).

this works fine:
```py
with BuildPart() as p:
    Box(25,10,10)
    with Locations((0,20)):
        Box(10,10,10)
    with BuildSketch() as s:
        Text('B',8,font="Arial")
    extrude(amount=15,mode=Mode.SUBTRACT)

exporter = Mesher()
exporter.add_shape(p.part)
exporter.write('3MF_test_2boxes_BCtext.3mf')
```
and this works too:
```py
with BuildPart() as p:
    Box(25,10,10)
    with Locations((0,20)):
        Box(10,10,10)
    with BuildSketch() as s:
        Text('B',8,font="Arial") #this works with below
        with Locations((5,0)): 
           Text('C',8,font="Arial") #this works with above
    extrude(amount=15,mode=Mode.SUBTRACT)
#omitted same exporter statements as above
```
This does NOT work:
```py
with BuildPart() as p:
    Box(25,10,10)
    with Locations((0,20)):
        Box(10,10,10)
    with BuildSketch() as s:
        Text('BC',8,font="Arial") #this does NOT work alone
    extrude(amount=15,mode=Mode.SUBTRACT)
#omitted same exporter statements as above
```
Neither does this:
```py
with BuildPart() as p:
    Box(25,10,10)
    with Locations((0,20)):
        Box(10,10,10)
    with BuildSketch() as s:
        Text('B C',8,font="Arial") #this does NOT work alone
    extrude(amount=15,mode=Mode.SUBTRACT)
#omitted same exporter statements as above
```

gumyr, 2024-01-07:
@jdegenstein would you please try this out again?  It works for me now with commit (cd834d1a16211e8b8613514deef7ecf8fbf791b7) but since fonts are platform dependent it still might be a problem.

jdegenstein, 2024-01-07:
Tested again on latest commit and can confirm this is resolved! Thanks everyone who contributed to the fix!

closed 2024-01-07 by jdegenstein