gumyr/build123d #317: VSCode Docker config - interested?

rhempel, 2023-09-10:
I have successfully installed build123d on a default python docker image fully automagically - just open the folder containing the .devcontainer folder and VSCode looks after:

- Pulling the Docker image
- Installing the required Linux libraries
- Creating a default build123d user with sudo access
- Pulling in build123d using a pip requirements file
- Installing the required VSCode plugins

You still have to run the build123d quickstart whenever you create or rebuild the Docker instance, and it takes a minute or so to install additional libraries, but it's pretty painless.

Build123d is now my go-to parametric modelling application - I was on the seemingly abandoned OpenSCAD and skipped CADQuery for build123d.

Great job putting this out there!

gumyr, 2023-09-10:
Happy to hear that you like using build123d.

Most build123d users are going to use https://github.com/bernhard-42/vscode-ocp-cad-viewer which can automatically install build123d.  Would you compare your approach to ocp_vscode installation?

gregrafferty, 2023-10-24:
I'd love to see anything that makes setting up easier, especially on an Apple Silicon Mac. I'm sure this wouldn't necessarily be completely plain sailing but a Docker image is my ideal. If I can mostly run a single command to get going that would be perfect. With build123d and CadQuery I keep getting tripped up on the install process.

fxdave, 2025-03-14:
After hours trying to make it work. I really just need a Dockerfile.

rhempel, 2025-03-14:
What host machine are you trying to make build123d work on, and what issue are you coming up against?

I have moved from Docker to PodMan but PodMan understands Docker files :-)

I can clean up my Dockerfile for build123 on the weekend and share it with you if you like .

fxdave, 2025-03-14:
That would be awesome! Thanks!

rhempel, 2025-03-16:
I'm not sure what your host machine is, and I would appreciate if we could walk through the install of this container so that I can see what you are doing, in case I can clean up the instructions ... I have moved to Podman from Docker but it should "just work" with either container manager.

My devcontainer repo is here:

https://github.com/rhempel/devcontainers

1. Clone the repo and follow the directions in the top level README.md to create the ubuntu-24.04-python base image.
2. Then right-click on the build123d-24.04 folder and choose "Open in VSCode"
3. VSCode will ask if you want to make a container in the bottom right corner, say yes
4. You may need to wait 10-20 minutes to build the container (on Windows) and possibly disable a corporate firewall if you have one

On my Linux box it builds much faster, not sure about apple silicon

Final step, install the ocp-viewer extension - it should figure out that it's connected to a container and try to communicate through port 3939 that I set up in the devcontainer already.

Maybe this is all more trouble than it's worth, but I find it helpful to not mess up my main development machine by creating containers - and if the container is good I just save it for restarting whenever I need it.

Hope this helps, and if you need any help, you can find me pretty easily on liked in and request a connect.