gumyr/build123d #318: Joints are lost when modifying a part

philipsd6, 2023-09-11:
As discussed in chat, here is a demonstration. The goal is to build a box in the same way as building a box with boards in real life, with butt joints (which may not be ideal, but whatever!)

```python
#!/usr/bin/env python3
import copy
from build123d import *
from ocp_vscode import *

set_defaults(reset_camera=Camera.CENTER)

board_width = 10
board_depth = 1 / 2
board_length = 20
z_points = [5, 15]
hole_radius = 1 / 16
hole_length = 3 / 2

with BuildPart() as board_builder:
    box = Box(board_width, board_depth, board_length)
    with Locations(box.faces().sort_by(Axis.Y).first):
        offset = board_width / 2 - board_depth / 2
        with Locations(*[(pt - board_length / 2, offset) for pt in z_points]):
            front_holes = CounterBoreHole(
                radius=hole_radius,
                counter_bore_radius=hole_radius * 4 / 3,
                counter_bore_depth=hole_radius * 3 / 2,
                depth=board_depth,
            )
    with Locations(box.faces().sort_by(Axis.X).last):
        with Locations(*[(pt - board_length / 2, 0) for pt in z_points]):
            side_holes = Hole(radius=hole_radius, depth=hole_length - board_depth) # [1]

    # Add common joints
    for i, hole in enumerate(front_holes):
        RigidJoint(
            label=f"back_hole_{i}",
            joint_location=hole.location * Pos(0, 0, board_depth),
        )
    for i, hole in enumerate(side_holes):
        RigidJoint(label=f"side_hole_{i}", joint_location=hole.location)

front = copy.copy(board_builder.part)
left = copy.copy(board_builder.part)
back = copy.copy(board_builder.part)
right = copy.copy(board_builder.part)

# I expect that after cutting an opening in the right board, I should still have my
# original joints, but if the below section is uncommented, the right board loses
# its joints and ends up occupying the same place as the front board.

# right -= Pos(board_depth / 2, 0, 0) * Box(
#     length=board_width * 2 / 3,
#     width=board_depth,
#     height=board_length * 1 / 3,
# )

# Programatically connect the joints...
for i in range(3):
    cur, nxt = [front, left, back, right][i : i + 2]
    for back_joint, side_joint in zip(
        [j for n, j in cur.joints.items() if n.startswith("back")],
        [j for n, j in nxt.joints.items() if n.startswith("side")],
    ):
        back_joint.connect_to(side_joint)

# Conceptually, I would add nails at this stage, but skipped for clarity.

# I can cut that opening in the right board after the joints are connected, but finding
# the place to cut is much more involved.

# right -= (
#    Plane(right.faces().sort_by(Axis.X).last.center())
#    * Rot(0, 0, 90)
#    * Pos(board_depth / 2, board_depth / 2, 0)
#    * Box(
#        length=board_width * 2 / 3,
#        width=board_depth,
#        height=board_length * 1 / 3,
#    )
#)

show(
    front, left, back, right,
    names=["front", "left", "back", "right"],
    render_joints=True,
    transparent=True,
)
```
My main issue is noted in the code comments -- the first one doesn't work; the second one does, but is more difficult.

[1] Note that if this joint is modified (say, by using `board_depth/2`) then the joint is "out in space" i.e. there is a gap between the joints. Since that is the case, why couldn't it be true that any joint could still exist, as long as the Part was still in existence, even if the original target of the joint on that Part was subtracted or modified? The joint should remain in the context of the Part as whole. So if the Part is moved, rotated, the joint goes along, and something like cutting off the side that a joint is supposed to attach to isn't an error per se, it just might be a mistake in design. 

As I said, I'm new to 3d coding, so this is just my newbie perspective, wdyt?

gumyr, 2023-09-19:
If an object with joints is modified within the BuildPart scope the joints will be re-applied to the object on exit but this isn't available outside of Builder mode.  It isn't really clear to me what to do here - the Joints might not be appropriate anymore.

Perhaps a method to easily copy joints from one object to another would be a partial solution to this request?

philipsd6, 2023-09-19:
Sure, that would be a partial solution... I did try using `copy.copy` & `copy.deepcopy` on the joints before modifying the parts in the algebra context, and then setting the `part.joints` value to the copy... but that doesn't work. The joints are put back in, but they are... wrong.

gumyr, 2024-11-08:
With commit 60029d81d571a2a6fad52cf45d5d7a4747c4955b this works now for both Algebra and Builder modes.
```py
class Pipe(BasePartObject):
    def __init__(self, length: float, od: float = 1.25 * IN, id: float = 1 * IN):
        profile = Circle(od / 2) - Circle(id / 2)
        pipe = extrude(profile, length)
        self.length = length
        super().__init__(pipe)
        RigidJoint(f"end0", self, joint_location=Location())
        RigidJoint(f"end1", self, joint_location=-Location((0, 0, length)))
        self.color = Color("Gainsboro")
        self.label = f"Pipe-{length/IN:0.3f}"

with BuildPart() as base_builder:
    Box(6 * IN, 6 * IN, 6 * IN)
    top_face = base_builder.faces().sort_by(Axis.Z)[-1]
    RigidJoint("top", joint_location=top_face.location_at(0.5, 0.5))
    with BuildSketch(base_builder.faces()):
        Circle(2 * IN)
    extrude(amount=-1 * IN, mode=Mode.SUBTRACT)
    base_builder.part.color = Color("SteelBlue")

base = base_builder.part

p = Pipe(1 * FT)
pipe_with_hole = p - Pos(Z=6 * IN) * Cylinder(0.25 * IN, 2 * IN, rotation=(90, 0, 0))
base.joints["top"].connect_to(pipe_with_hole.joints["end0"])
```
[image]

closed 2024-11-08 by gumyr