gumyr/build123d #319: 2D objects have a non-null volume

Jojain, 2023-09-11:
To reproduce : 
```py
>>> from build123d import *
>>> f = Face.make_rect(10,20)
>>> f.volume
199.99999999999997
```

Same with 2D shells

jdegenstein, 2023-09-19:
From the OCCT docs: "
<html>
<body>
<div class="memproto" style="font: bold 14px / 22px Roboto, sans-serif; border-top: 1px solid rgb(168, 184, 217); border-left: 1px solid rgb(168, 184, 217); border-right: 1px solid rgb(168, 184, 217); padding: 6px 0px; color: rgb(37, 53, 85); text-shadow: rgba(255, 255, 255, 0.9) 0px 1px 1px; background-color: rgb(223, 229, 241); box-shadow: rgba(0, 0, 0, 0.15) 5px 5px 5px; border-top-right-radius: 4px; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">

Standard_Real GProp_GProps::Mass | ( |   | ) | const
-- | -- | -- | -- | --

</div><div class="memdoc" style="font: 400 14px / 22px Roboto, sans-serif; border-width: 0px 1px 1px; border-bottom-style: solid; border-bottom-color: rgb(168, 184, 217); border-left-style: solid; border-left-color: rgb(168, 184, 217); border-right-style: solid; border-right-color: rgb(168, 184, 217); padding: 6px 10px 2px; background-image: url(&quot;nav_g.png&quot;); background-repeat: repeat-x; background-color: rgb(255, 255, 255); border-bottom-left-radius: 4px; border-bottom-right-radius: 4px; box-shadow: rgba(0, 0, 0, 0.15) 5px 5px 5px; color: rgb(0, 0, 0); letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><p style="font: 400 14px / 22px Roboto, sans-serif;">Returns the mass of the current system. If no density is attached to the components of the current system the returned value corresponds to :</p><ul style="overflow: hidden;"><li>the total length of the edges of the current system if this framework retains only linear properties, as is the case for example, when using only the LinearProperties function to combine properties of lines from shapes, or</li><li>the total area of the faces of the current system if this framework retains only surface properties, as is the case for example, when using only the SurfaceProperties function to combine properties of surfaces from shapes, or</li><li>the total volume of the solids of the current system if this framework retains only volume properties, as is the case for example, when using only the VolumeProperties function to combine properties of volumes from solids. Warning A length, an area, or a volume is computed in the current data unit system. The mass of a single object is obtained by multiplying its length, its area or its volume by the given density. You must be consistent with respect to the units used.</li></ul></div>
</body>
</html>

https://dev.opencascade.org/doc/refman/html/class_g_prop___g_props.html#a1796c83e8b5c53b959e41c55e87112a5

Unfortunately the property of dim is listed as protected, so does not look like that is accessible from this class (I am sure it is available elsewhere).

https://github.com/CadQuery/OCP/blob/9ab13441296788f8b04ac2d0e8a90f4533587fa6/opencascade/GProp_GProps.hxx#L99

As it stands this is "not a bug" as far as OCCT is concerned, but arguably should be improved at the build123d level.

jdegenstein, 2024-01-19:
Initially I thought that returning `None` was the obvious choice, but there is at least one method that depend on volume always returning a float. Here is what I have so far:
```py
class Shape ...
    @property
    def volume(self) -> float:
        """volume - the volume of this Shape"""
        # when density == 1, mass == volume
        if self.solids():
            return Shape.compute_mass(self)
        else:
            return 0.0
```
@Jojain is this good enough? Here are the tests results when the last line is `return None`

https://github.com/jdegenstein/build123d/actions/runs/7579670156/job/20644284638

Jojain, 2024-01-19:
Returning 0 is fine I think

mnesarco, 2024-01-19:
Hi, returning 0 is ok, but maybe building the solids list is a costly check.

jdegenstein, 2024-01-19:
@mnesarco suggestions to avoid this are welcome, just trying to do this in a relatively simple way for now.

mnesarco, 2024-01-19:
I just wonder why not this:

```python
class Shape ...
    @property
    def volume(self) -> float:
        """volume - the volume of this Shape"""
        properties = GProp_GProps()
        BRepGProp.VolumeProperties_s(self.wrapped, properties)
        return properties.Mass()

```        

Maybe it was already tried with bad results?

jdegenstein, 2024-01-19:
Please review the PR https://github.com/gumyr/build123d/pull/488 I made related to this issue

jdegenstein, 2024-01-31:
Closed by https://github.com/gumyr/build123d/pull/488

closed 2024-01-31 by jdegenstein