gumyr/build123d #327: Intersections on planes and axis

Jojain, 2023-09-21:
It would be nice to have more features on the "geometry" objects and not rely only on the topo object.
Nice improvement would be :
- Intersections of plane between plane, returns an axis, which start point could be the projected mid point of the origin of the two plane onto the axis
- Intersections of axis with axis, it would return a point. The axis would need to be in the same plane or if not maybe the intersection can be return in the mid plane of both axis.
- It can be nice to have a function to compute a mid axis given two axis.

gumyr, 2024-06-19:
@Jojain I'm trying to complete this feature now - I need more info on the 3rd point.  Would this case only apply to two Axes that are parallel?

Jojain, 2024-06-20:
No, the axis can be in whatever orientation, 2 axis can always be considered in the same plane, with potentially one of the axis offset in the Z direction of the common plane. In such case the start point of the mid axis is the intersection of the two axis in the common plane projected in the mid plane. And the mid axis direction is the mid vector of the axis direction.

[image]

Would yield: 
[image]

Then the case when axis are parallel is a special case where the start point is the mid point of both axis start point as you won't be able to find an intersection

gumyr, 2024-06-20:
Thanks for the clarification. This mid axis feature seems to be very specialized - when would this feature be used?

Jojain, 2024-06-22:
I don't think its very specialized, I think it's a basic geometric tool. Basically it does in one function call a mid vector computation as well as a mid point one. If you will it's not needed no need to bother with that though, intersections are the priority !

gumyr, 2024-06-23:
I've split the mid axis feature out to a separate Issue so I can close this issue.  Now all geometric objects (Vector, Location, Axis, Plane) can intersect each other and at least is some a basic way intersect Shapes.  However, there are many possibilities when doing Shape intersections so I expect new Shape functionality will be required to cover off all of the possibilities.

closed 2024-06-23 by gumyr