gumyr/build123d #328: Intersect Everything

gumyr, 2023-09-22:
As mentioned in Issue #327 it should be possible to intersect the geometric objects with each other but also the topological objects. Much of this functionality is already present but it should be reorganized to be consistent across the whole project.  It may be desirable to have specific names for some operations, for example a `section` is just the intersection of a `Plane` and a `Shape` so the logic should be in the `intersect` method(s) wrappers for well understood names.

A survey of the related existing intersection related methods needs to be done to assess what is missing.

gumyr, 2023-09-25:
The existing intersection related methods are as follows:

```py
class Plane:
    def find_intersection(self, axis: Axis) -> Union[Vector, None]:
        """Find intersection of axis and plane"""

class Shape:
    def intersect(self, *to_intersect: Shape) -> Self:
        """Intersection of the positional arguments and this Shape.

        Args:
            toIntersect (sequence of Shape): shape to intersect

        Returns:

        """

    def faces_intersected_by_axis(
        self,
        axis: Axis,
        tol: float = 1e-4,
    ) -> ShapeList[Face]:
        """Line Intersection

        Computes the intersections between the provided axis and the faces of this Shape

        Args:
            axis (Axis): Axis on which the intersection line rests
            tol (float, optional): Intersection tolerance. Defaults to 1e-4.

        Returns:
            list[Face]: A list of intersected faces sorted by distance from axis.position
        """

    def find_intersection(self, axis: Axis) -> list[tuple[Vector, Vector]]:
        """Find point and normal at intersection

        Return both the point(s) and normal(s) of the intersection of the axis and the shape

        Args:
            axis (Axis): axis defining the intersection line

        Returns:
            list[tuple[Vector, Vector]]: Point and normal of intersection
        """

class Compound:
    def do_children_intersect(
        self, include_parent: bool = False, tolerance: float = 1e-5
    ) -> tuple[bool, tuple[Shape, Shape], float]:
        """Do Children Intersect

        Determine if any of the child objects within a Compound/assembly intersect by
        intersecting each of the shapes with each other and checking for
        a common volume.

        Args:
            include_parent (bool, optional): check parent for intersections. Defaults to False.
            tolerance (float, optional): maximum allowable volume difference. Defaults to 1e-5.

        Returns:
            tuple[bool, tuple[Shape, Shape], float]:
                do the object intersect, intersecting objects, volume of intersection
        """

    def intersect(self, *to_intersect: Shape) -> Compound:
        """Construct shape intersection

        Args:
          *to_intersect: Shape:

        Returns:

        """

class Edge:
    def intersections(
        self, plane: Plane, edge: Union[Axis, Edge] = None, tolerance: float = TOLERANCE
    ) -> list[Vector]:
        """intersections

        Determine the points where a 2D edge crosses itself or another 2D edge

        Args:
            plane (Plane): plane containing edge(s)
            edge (Edge): curve to compare with
            tolerance (float, optional): defines the precision of computing the intersection points.
                 Defaults to TOLERANCE.

        Returns:
            list[Vector]: list of intersection points
        """
```
The `Shape.intersect` returns `Self` which isn't going to always be true.  An `Edge` intersected with a `Solid` could result in multiple `Edges` which would likely be a `Compound`.

To avoid circular imports any methods added to the geometry classes must be monkey-patched in topology.py.

gumyr, 2023-09-25:
A table of topology/geometry intersections and outcomes:
[image]

It seems like the complex method outputs will mess with typing but these are the possibilities.

gumyr, 2025-03-03:
See: https://forum.freecad.org/viewtopic.php?t=33078 about the use of `GeomAPI_ExtremaCurveCurve` for intersection detection.

gumyr, 2025-03-03:
Also: https://tracker.dev.opencascade.org/view.php?id=30217

jwagenet, 2025-03-03:
```python
r = 23
with BuildLine() as line:
    c1 = CenterArc((0, 0), r + 10, start_angle=0, arc_size=360)
    c2 = CenterArc((20, 0), r + 5, start_angle=0, arc_size=360)
    x = c1.edge().intersect(c2.edge())
    print(f"control r: {r}, intersections: {x}")
    print(f"minimum r should be > {(20 - (10 + 5)) / 2}")
show(x, line)
```
An intersection example that fails but shouldn't. Intersection(s) `x` has two results for integers `4 <= r <= 22` and `None` for smaller or larger. The expected range is `2.5 < r <= math.inf`.

jwagenet, 2025-08-14:
> A table of topology/geometry intersections and outcomes: [image]

I realize this table is quite old, regarding the expected output types:
* Is there a preference for returning Vertex vs Vector? (I've gotten both in some testing) I suppose a distinction could be made whether a is topological or geometric.
* Is it realistic/desired to return a Compound/Curve/Sketch/Part or Wire/Shell as opposed to a ShapeList[Vertex | Edge | Face | Solid]?

gumyr, 2025-08-15:
Thanks for looking into this!

> Is there a preference for returning Vertex vs Vector? (I've gotten both in some testing) I suppose a distinction could be made whether a is topological or geometric. This applies to returning Axis/Plane vs Edge/Face as well

A Vertex should be returned if that is the `Shape` at the intersection.  For example, two Edges might intersect at a Vertex; however, since an Axis doesn't have any Vertices the intersection of two Axes could only result in a Vector (or an Axis). Returning all of the appropriate Shapes can be a bit of a challenge as there are different OCCT mechanisms to extract different types of objects.

> Is it realistic/desired to return a Compound/Curve/Sketch/Part or Wire/Shell as opposed to a ShapeList[Vertex | Edge | Face | Solid]?

The big refactor that happened at the end of 2024 has changed the situation.  In the past returning a Compound seemed that the most convenient thing to do. However, the code restructuring was done to break up one massive module (~10K lines) into the entire `topology` sub-directory of modules.  A core part of this is a new import structure as follows:

- geometry > shape_core > zero_d > one_d > two_d > three_d > composite

therefore, one_d (i.e. Edge and Wire) doesn't know about Compound and can't use it.  In place of Compound a ShapeList is used where more than one Shape must be returned. For consistency, it probably makes sense that all of the intersections return ShapeLists.  Users are able to use ShapeList.vertices() (etc.) to extract Shapes from the ShapeLists in much the same way they would extract objects from Compounds so there should be very little practical difference.

jwagenet, 2025-08-29:
Currently, geometry intersections with `Location`s only return a `Location` if the orientation for the non-`Vector` object matches the `Location` (and co-origin with `Plane`), otherwise returns None. I see value in confirming shared orientation between objects, but returning None when eg `Location` is coplanar with `Plane` but not oriented seems wrong. Suggested changes:

- `Location` intersecting `Location | Axis | Plane` returns `Location` if coincident and co-oriented to each other regardless of position along infinite shape, and `Vector` if not co-oriented  
-  Uncertain about `Location` intersecting `Vector` since the latter lacks orientation

Similarly, intersecting coplanar `Plane`s returns None if they don't share origin, but I'd suggest this case returns `self: Plane` instead.

jwagenet, 2025-09-02:
[image]

Yea, the suggestion for `Axis` intersection, where the left `Location` is on axis, but off orientation and would return a `Vector` intersection and the right is on orientation to return `Location`. Neither are on origin.

[image]

The suggestion for `Plane` intersection, where the right `Location` is on plane, but off orientation and would return a `Vector` intersection and the left is on orientation to return `Location`. Neither are on origin.

Currently, because of the co-origin requirement, non of those cases return an object, which seems wrong to me. However, intersecting with a `Location` seems like an unusual case. Perhaps in your mind, a `Location` should only have a valid intersection with the `location` of a `Plane` or `Axis`.

gumyr, 2025-09-02:
I agree with all of this.  Returning either a Vector or a Location is a good idea, it tells the user more about what is happening.  I'm still not sure why one would do a Location/Plane intersection so it's difficult to know what the correct answer is here.

gumyr, 2025-09-04:
I don't recall if the code still has it but for a while I restricted intersect to a single value to avoid this problem - what is the user intention? I think `A.intersect([B, C]) = [A ∩ B ∩ C]` needs to follow the math (which matches how Python set intersection works):
```py
>>> set_a = {"a"}
>>> set_b = {"a", "b", "c"}
>>> set_c = {"a", "d", "e"}
>>> set_d = {"b", "c"}
>>> set_a.intersection(set_b,set_c)
{'a'}
>>> set_a.intersection(set_b, set_d)
set()
```
However, maybe a matrix of intersection results (i.e. `list` of `ShapeList` as returned by `group_by`) would be valuable?

jwagenet, 2025-09-16:
To summarize discussed expected behavior from the `intersect` method for different object types:

Geometry Objects
* `Vector`, `Location`, `Axis`, `Plane`
* Geometry/geometry intersections produce geometry, Geometry/`Shape` produces `Shape`
* Accept 1 arg and returns a single object (or `None`)
* `Location` is a returns a `Vector`, except when orientation is shared with intersection target

Shape Objects
* Accept n args and returns a `ShapeList[Shape]` (or `None`) even if there is a single intersection
* `Compound` output is replaced with `ShapeList[Vertex | Edge | Face | Solid]`
* Due to topology refactor, `Shape`s need an `intersect` method at each dim level (`Vertex`, `Mixin1D`, etc)
* `A.intersect(B, C) = [A ∩ B ∩ C]` is expected output (all intersect all). If user wants only A intersect all, something like `set([A.intersect(x) for x in [B, C]])` required.