gumyr/build123d #335: Sweep does not properly remove sketch from pending

jdegenstein, 2023-09-29:
I think sweep is not properly removing a sketch from pending objects, see below example:
```py
with BuildLine() as l:
    m0 = Line((0,0),(30,0))
    m1 = JernArc(m0@1,m0%1,75+30,45)
    m2 = Line(m1@1,m1@1+(m1%1)*40)

with BuildLine() as l2:
    n1 = Line(m2@1,m2@1+(-70*(m2%1).Y,70*(m2%1).X))
    n2 = Line(n1@0,n1@0+(n1%0)*-40)
    n3 = JernArc(n2@1,n2%1,75+30,45)
    n4 = Line(n3@1,n3@1+(n3%1)*30)

with BuildPart() as p:
    with BuildSketch(Plane.YZ) as s:
        Circle(60/2)
    sweep(s.sketch,l.line)

    with BuildSketch(Plane.YZ.offset((n4@1).X)) as s3:
        Circle(60/2)
    sweep(s3.sketch,l2.line)

    with BuildSketch(Plane.YZ.offset((n4@1).X)) as s5:
        Circle(85/2)
    extrude(amount=-20) #results in incorrect behavior of screenshot with extrusion of both s5.sketch and s.sketch
    #extrude(s5.sketch,amount=-20) #works properly
```

[image]

jdegenstein, 2023-10-06:
Maybe this is intended behavior? I see that pending_faces is not altered when sections are manually specified (as I did above in the two sweeps).

gumyr, 2023-10-10:
Yes this is intended.  When a face is passed as a parameter there is no attempt to figure out if it's one of the pending faces and remove it as this would be a dangerous side effect in some situations.  When the user doesn't provide a face the Builder is under control and should always clean up after itself.

closed 2023-10-10 by gumyr