gumyr/build123d #338: Plane does not seem to actually adopt results of modifications

jdegenstein, 2023-10-08:
If you substitute `split(bisect_by=pln2 ...` with `pln` the result is not correct. I have had this issue before and I am really not sure what I am doing wrong? I constructed `pln2` by printing out `pln`'s origin/x_dir/z_dir and creating a new plane with the same values as `pln`.
```py
# %%
from build123d import *
from ocp_vscode import *
set_port(3939)
set_defaults(reset_camera=Camera.CENTER,ortho=True)
densa = 7800/1e6 #carbon steel density g/mm^3
densb = 2700/1e6 #aluminum alloy
densc = 1020/1e6 #ABS
ms = Mode.SUBTRACT
# %%

with BuildPart() as p:
    with BuildLine() as l:
        m1 = Line((0,0),(3.75/2,0))
        m2 = JernArc(m1@1,m1%1,.25,-90)
        m3 = Line(m2@1,m2@1+(0,-4+.125+.25))
    z1 = make_brake_formed(.125,4-.375,line=l.line)

    with BuildPart() as p01:  
        with BuildLine(Plane.YZ) as l2:
            n1 = JernArc((0,0),(0,-1),.25,-90)
            n2 = Line(n1@1,n1@1+(-1+.125+.25,0))
        z2 = make_brake_formed(.125,3.75/2,line=l2.line)
        fillet(z2.edges().filter_by(Axis.Z).group_by(Axis.Y)[0].sort_by(Axis.X)[-1],.25)
    
    with BuildPart(mode=Mode.PRIVATE) as p02: 
        with BuildLine(Plane.XZ.offset(0)) as l3:
            p1 = JernArc((0,0),(0,-1),.25,-90)
            p2 = Line(p1@1,p1@1+(-1.5+.125+.25,0))
        z3=make_brake_formed(.125,2,line=l3.line)
        fillet(z3.edges().filter_by(Axis.Z).group_by(Axis.X)[0],.25)
    with Locations((4.5/2-.125,-4+.125+2)):
        add(p02.part)
    with Locations((2.5/2,-4+.125+1)):
        Hole(.5/2)
    pln = Plane.XY.offset(4-.375).rotated((37,0,0))
    print(pln)
    pln.origin = pln.origin + (0,-.375+.125,0)
    print(pln)

    pln2 = Plane((0.0, -0.375+.125, 3.625), (1.00, 0.00, 0.00), (0.0, -0.6018150231520483, 0.7986355100472928))
    split(bisect_by=pln2,keep=Keep.BOTTOM)
    with Locations(Plane.XZ):
        with Locations((0,4-.375-2)):
            Hole(2.5/2)
    fillet(p.part.edges().filter_by(Axis.X).group_by(Axis.Y)[0].sort_by(Axis.Z)[-1],1)
    mirror(about=Plane.YZ)

print(f"\npart mass = {p.part.scale(IN).volume*densb/453.592}") #.472 lbs, correct is .472 lbs +/- 0.002 lbs
```

gumyr, 2023-10-10:
Fixed.  When using the origin setter an important step was missing so the gp_Pln wasn't being updated.

closed 2023-10-10 by gumyr

jdegenstein, 2023-10-11:
Thanks Roger!