gumyr/build123d #342: GridLocations align parameter only works with tuple form

philipsd6, 2023-10-10:
The GridLocations type is documented to support the align parameters as: `Union[Align, tuple[Align, Align]]'` However only the tuple form is working.

```python
with BuildPart() as thing:
    with BuildSketch() as sketch:
        grid = Rectangle(20, 20)
        with GridLocations(2, 2, 3, 3, align=Align.MIN) as grid_locs:
            holes = Rectangle(1, 1, mode=Mode.SUBTRACT)
    extrude(amount=5)
```
Results in:
```python-traceback
   1037 align_offset = []
   1038 for i in range(2):
-> 1039     if align[i] == Align.MIN:
   1040         align_offset.append(0.0)
   1041     elif align[i] == Align.CENTER:

TypeError: 'Align' object is not subscriptable
```

jdegenstein, 2023-10-11:
Can you test and close this issue if it is fixed for you? My PR was merged.

philipsd6, 2023-10-13:
Yup, seems to be fixed!

closed 2023-10-13 by philipsd6