gumyr/build123d #344: Entering BuildPart context without explicitly providing a plane clears the plane of parent context

IsawU, 2023-10-11:
`build123d` version: `0.1.dev1033+g0b30011`

I came across this issue where entering `BuildPart()` context without specifying the plane removes the plane in parent context for any subsequent calls in the parent context.

See the following snippet:

```python
with BuildPart(Plane.XZ) as part:
    Cylinder(1, 3)

    with BuildPart():   # No plane provided here.
        Box(1, 1, 4)    # This box is not important.

    # The Plane.XZ is lost here.

    Cylinder(1, 3)  # This cylinder has a different orientation from the first one.
```

The second `Cylinder()` is rotated by 90° as if created on a `Plane.XY`.
Adding any plane to the nested `BuildPart()` avoids this and makes both the `Cylinder()`s coincide.

Some more observations:

- `BuildSketch()` also causes the plane to be lost.
- `BuildLine()` does not trigger this. _This seems to be because `BuildLine()` has an explicit `workplane: Union[Face, Plane, Location] `**`= Plane.XY`** in `__init__()` declaration._
- Only the parent context's plane is lost:

```python
with BuildPart(Plane.XZ) as part:
    Cylinder(1, 3)

    with BuildPart(Plane.ZX):
        Box(1, 1, 4)

        with BuildSketch():     # No plane provided here.
            Rectangle(1, 1)

        # The Plane.ZX is lost here.

        Box(1, 1, 4)    # This box has a different orientation from the first one.

    # The Plane.XZ is still in effect here.

    Cylinder(1, 3)  # This cylinder has the same orientation as thefirst one.
```

I looked through the source and I found the following:<br>_I'm linking to a commit that I'm certain is different from my version, but the relevant code looks to be the same. I couldn't figure out what commit `0.1.dev1033+g0b30011` is on._

If this line runs, the plane is lost. [`build_common:221`](https://github.com/gumyr/build123d/blob/b19bb81047001e5eeb0aa1bcacd0b99c8009b398/src/build123d/build_common.py#L221)

I can add `self.workplanes = [Plane.XY]` or `= self.builder_parent.workplanes` here: [`build_common:203`](https://github.com/gumyr/build123d/blob/b19bb81047001e5eeb0aa1bcacd0b99c8009b398/src/build123d/build_common.py#L203) to get either `Plane.XY` or inherit parent context's plane.
But any value assigned here renders the `if` linked in the previous paragraph completely redudndant.

Perhaps the fact that `self.workplanes` is not set, only `self.workplanes_context`, is the main cause. I tried the following which also fixes the issue and does not seem hijack the surrounding logic:

```python
        # If there are no workplanes, create a default XY plane
        if self.workplanes:
            self.workplanes_context = WorkplaneList(*self.workplanes).__enter__()
        else:
            self.workplanes_context = WorkplaneList(Plane.XY).__enter__()
            self.workplanes = self.workplanes_context.workplanes
```

gumyr, 2023-10-15:
Thanks for raising this issue - it appears to be somewhat serious.  I can see from the log:
```bash
INFO:root:Starting issue 344 debug
INFO:build123d:None context requested by None
INFO:build123d:Entering BuildPart with mode=Mode.ADD which is in different scope as parent
INFO:build123d:WorkplaneList is pushing 1 workplanes: [Plane(o=(0.00, 0.00, 0.00), x=(1.00, 0.00, 0.00), z=(0.00, -1.00, 0.00))]
INFO:build123d:LocationList is pushing 1 points: [(p=(0.00, 0.00, 0.00), o=(-0.00, 0.00, -0.00))]
INFO:build123d:BuildPart context requested by Cylinder
INFO:build123d:Completed integrating 1 object(s) into part with Mode=Mode.ADD
INFO:build123d:BuildPart context requested by None
INFO:build123d:BuildPart context requested by None
INFO:build123d:BuildPart context requested by None
INFO:build123d:Entering BuildPart with mode=Mode.ADD which is in same scope as parent
INFO:build123d:WorkplaneList is pushing 1 workplanes: [Plane(o=(0.00, 0.00, 0.00), x=(1.00, 0.00, 0.00), z=(0.00, 0.00, 1.00))]
INFO:build123d:LocationList is pushing 1 points: [(p=(0.00, 0.00, 0.00), o=(-0.00, 0.00, -0.00))]
INFO:build123d:BuildPart context requested by Box
INFO:build123d:Completed integrating 1 object(s) into part with Mode=Mode.ADD
INFO:build123d:Completed integrating 1 object(s) into part with Mode=Mode.ADD
INFO:build123d:Exiting BuildPart
INFO:build123d:BuildPart context requested by Cylinder
INFO:build123d:Completed integrating 1 object(s) into part with Mode=Mode.ADD
INFO:build123d:LocationList is popping 1 points
INFO:build123d:WorkplaneList is popping 1 workplanes
INFO:build123d:Exiting BuildPart
```
that there should be a `WorkplaneList is popping 1 workplanes` when the second `BuiltPart` exits but it's missing. More investigation required...

IsawU, 2023-10-16:
I might have the culprit now.

The `Builder.__exit__()` gets a list of `self.exit_workplanes`, but then never uses them, instead it calls exit on each `self.workplane` which in this case is an empty list (Except for `BuildLine()` which has en explicit `Plane.XY`).

[build_common:248](https://github.com/gumyr/build123d/blob/b19bb81047001e5eeb0aa1bcacd0b99c8009b398/src/build123d/build_common.py#L248)

```python
        self.exit_workplanes = WorkplaneList._get_context().workplanes

        # Now that the object has been transferred, it's save to remove any (non-default)
        # workplanes that were created then exit
        if self.workplanes:  # NOTE: Changing this to self.exit_workplanes also fixes the issue.
            self.workplanes_context.__exit__(None, None, None)
```

gumyr, 2023-12-11:
Closed with commit b2eb1fb1b7ef9b6a603af467ad0a558e0a7b6219 .  The problem was that the Builder's `workplanes` list was not being set when default workplanes are used thus causing the workplane pop at exit to not happen.  Thanks for bringing it up and looking at it in depth.
 
[image]

closed 2023-12-11 by gumyr