gumyr/build123d #348: OCCT / OCP Message Output (STEP Writer)

jdegenstein, 2023-10-16:
This is not a critical issue, but I investigated fixes for the following behavior:
```py
In [1]: from build123d import *
In [4]: zz = Box(1,1,1)
In [5]: zz.export_step("box111_g.step")

*******************************************************************
******        Statistics on Transfer (Write)                 ******

*******************************************************************
******        Transfer Mode = 0  I.E.  As Is       ******
******        Transferring Shape, ShapeType = 0                      ******
** WorkSession : Sending all data
 Step File Name : box111_g.step(350 ents)  Write  Done
Out[5]: <IFSelect_ReturnStatus.IFSelect_RetDone: 1>
```
Rough prototype, which disables the transfer statistics:
```py
In [1]: zz = Box(1,1,1)
In [2]: from OCP import Message
In [3]: a = Message.Message.DefaultMessenger_s()
In [4]: mg = Message.Message_Gravity(3) #using the actual enums would be better
In [5]: for printer in a.Printers():
   ...:     printer.SetTraceLevel(mg)
In [6]: zz.export_step("box111_h.step")
Out[7]: <IFSelect_ReturnStatus.IFSelect_RetDone: 1>
```
I am not sure if this can be done globally for all of build123d, or alternatively within the `export_step` function itself.

gumyr, 2024-03-21:
Yes for the `export_step` function but not for the `Shape.export_step` method.  Should we close this issue or wait until the method is deprecated?

gumyr, 2024-03-23:
All of the `Shape.export` methods are now deprecated and this issue is fixed with the functions so I'll close it now.

closed 2024-03-23 by gumyr