gumyr/build123d #349: sweeping along a helix twists the 2d shape

yshui, 2023-10-18:
code:

```python
with BuildPart() as screw:
    with BuildLine():
        t = Helix(4, 30, 6.35)
    with BuildSketch(Plane(origin=t @ 0, z_dir = t % 0)):
        Rectangle(2, 2)
    sweep()
```

I expect the square to rotate around the axis of the helix, i.e. perpendicular to the helix. but instead the square also gets twisted around a axis that's tangential to the helix:

[image]

jdegenstein, 2023-10-18:
I believe this is expected behavior, here is how you can "fix" it by adding `is_frenet=True` to `sweep()`

[image]

yshui, 2023-10-18:
Thanks!

Can you explain a bit why it twists, and what is frenet?

jdegenstein, 2023-10-18:
There is a good writeup on the FreeCAD docs (build123d uses the same OCCT CAD core as FreeCAD, so typically most of the same considerations apply) https://wiki.freecad.org/Part_Sweep#Frenet

closed 2023-10-18 by yshui