gumyr/build123d #350: Mesher() can not handle mesh files with interior cavities/holes

jdegenstein, 2023-10-20:
When importing STL files using `Mesher()` a user reported issues with `Mesher()` closing unexpectedly. This simple cylinder STL works fine (see attached) [test_STLs.zip](https://github.com/gumyr/build123d/files/13057368/test_STLs.zip) 
[image]

```py
importer = Mesher()
stl = importer.read("cyl.stl")
stl = stl[0]
stl -= Box(10,10,10)
```
This STL with a cavity that is completely internal to the cylinder does NOT work, and returns the following error message:
```py
importer = Mesher()
stl = importer.read("cyl_w_rect_cavity.stl")
stl = stl[0]
stl -= Box(10,10,10)
```
```py
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Untitled-1 in line 5
      16 # %%
      19 importer = Mesher()
----> 20 stl = importer.read("cyl_w_rect_cavity.stl")
      21 stl = stl[0]
      22 stl -= Box(10,10,10)

File c:\Users\someuser\.conda\envs\ocp_vscode\Lib\site-packages\build123d\mesher.py:491, in Mesher.read(self, file_name)
    489 shapes = []
    490 for mesh in self.meshes:
--> 491     shape = self._get_shape(mesh)
    492     shape.label = mesh.GetName()
    493     # Extract color

File c:\Users\someuser\.conda\envs\ocp_vscode\Lib\site-packages\build123d\mesher.py:459, in Mesher._get_shape(self, mesh_3mf)
    457 shape_obj = Shell(occ_shell)
    458 if shape_obj.is_manifold:
--> 459     solid_builder = BRepBuilderAPI_MakeSolid(occ_shell)
    460     shape_obj = Solid(solid_builder.Solid())
    462 return shape_obj

TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. OCP.BRepBuilderAPI.BRepBuilderAPI_MakeSolid()
    2. OCP.BRepBuilderAPI.BRepBuilderAPI_MakeSolid(S: OCP.TopoDS.TopoDS_CompSolid)
    3. OCP.BRepBuilderAPI.BRepBuilderAPI_MakeSolid(S: OCP.TopoDS.TopoDS_Shell)
    4. OCP.BRepBuilderAPI.BRepBuilderAPI_MakeSolid(S1: OCP.TopoDS.TopoDS_Shell, S2: OCP.TopoDS.TopoDS_Shell)
    5. OCP.BRepBuilderAPI.BRepBuilderAPI_MakeSolid(S1: OCP.TopoDS.TopoDS_Shell, S2: OCP.TopoDS.TopoDS_Shell, S3: OCP.TopoDS.TopoDS_Shell)
    6. OCP.BRepBuilderAPI.BRepBuilderAPI_MakeSolid(So: OCP.TopoDS.TopoDS_Solid)
    7. OCP.BRepBuilderAPI.BRepBuilderAPI_MakeSolid(So: OCP.TopoDS.TopoDS_Solid, S: OCP.TopoDS.TopoDS_Shell)

Invoked with: <OCP.TopoDS.TopoDS_Compound object at 0x000001AA1F51B2B0>
```
Conversely, the STL with a rectangular thru-hole also does not work during the boolean operation, because `Mesher` returns a Shell (instead of a Solid, as I believe it should):
```py
importer = Mesher()
stl = importer.read("cyl_w_rect_hole.stl")
stl = stl[0]
stl -= Box(10,10,10)
```
```py
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Untitled-1 in line 7
      20 stl = importer.read("cyl_w_rect_hole.stl")
      21 stl = stl[0]
----> 22 stl -= Box(10,10,10)
     26 #print(f"\npart mass = {p.part.volume*densa}")
     27 classes = (BuildPart, BuildSketch, BuildLine) #for OCP-vscode

File c:\Users\someuser\.conda\envs\ocp_vscode\Lib\site-packages\build123d\topology.py:1688, in Shape.__sub__(self, other)
   1685 others = other if isinstance(other, (list, tuple)) else [other]
   1687 if not all([type(other)._dim == type(self)._dim for other in others]):
-> 1688     raise ValueError(
   1689         f"Only shapes with the same dimension can be subtracted "
   1690         f"not {type(self).__name__} and {type(other).__name__}"
   1691     )
   1693 new_shape = None
   1694 if self.wrapped is None:

ValueError: Only shapes with the same dimension can be subtracted not Shell and Box
```
The type of `stl[0]` is `build123d.topology.Shell`

jdegenstein, 2023-10-20:
Further information here: https://dev.opencascade.org/doc/refman/html/class_b_rep_builder_a_p_i___make_solid.html#details about how to create holes in Solids

gumyr, 2024-01-09:
Fixed now:
[image]

closed 2024-01-09 by gumyr