gumyr/build123d #362: Cannot access local variable 'offset_wire' where it is not associated with a value

gsingh93, 2023-10-30:
I'm getting the error "Cannot access local variable 'offset_wire' where it is not associated with a value" when running the following code:
```python
from build123d import *
from ocp_vscode import *

with BuildSketch() as s:
    Text("hello", font_size=10)
    offset(amount=-1)

show_all()
```

Output:
```
Traceback (most recent call last):
  File "/Users/gsgx/test.py", line 6, in <module>
    offset(amount=-1)
  File "/Users/gsgx/code/build123d/src/build123d/operations_generic.py", line 588, in offset
    outer_wire = face.outer_wire().offset_2d(amount, kind=kind)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/gsgx/code/build123d/src/build123d/topology.py", line 907, in offset_2d
    offset_edges = offset_wire.edges()
                   ^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'offset_wire' where it is not associated with a value
```

Even if there is some reason this example fundamentally shouldn't work, an `UnboundLocalError` usually indicates some deeper issue.

Changing `font_size` to something larger like 100 makes the error go away.

Tested on commit 159ff7d9c25dc2872e24ff6777fafddfdc51be9a.

snoyer, 2023-10-30:
Looks like there is a branch that leads to `offset_wire` not being initialized in here: https://github.com/gumyr/build123d/blob/dev/src/build123d/topology.py#L857C5-L865
```py
        if isinstance(obj, TopoDS_Compound):
            for i, el in enumerate(Compound(obj)):
                offset_wire = Wire(el.wrapped)
                if i >= 1:
                    raise RuntimeError("Multiple Wires generated")
```
this wants exactly one wire and will raise if it find more that 1, but the case of 0 wires is not checked and the code will carry on with no `offset_wire` variable

gumyr, 2023-11-04:
Thanks for raising this and for the help @snoyer, the problem was indeed having no code to detect when no offsets were created (an empty Compound returned).  A RuntimeError is generated now.

closed 2023-11-04 by gumyr